thingsboard/thingsboard-gateway #1175: [BUG]fromCoils() got an unexpected keyword argument 'wordorder'"

struggle3, 2023-09-10:
**Describe the bug**
1、使用tb-gateway 3.3读取 function code 1

**Connector name (If bug in the some connector):**
[e.g.Modbus TCP]

**Error traceback (If available):**
```
""2023-09-10 22:19:29" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 92 - fromCoils() got an unexpected keyword argument 'wordorder'"
Traceback (most recent call last):
  File "C:\Users\ye\AppData\Local\Programs\Python\Python39\lib\site-packages\thingsboard_gateway\connectors\modbus\bytes_modbus_uplink_converter.py", line 63, in convert
    decoder = BinaryPayloadDecoder.fromCoils(coils, byteorder=endian_order, wordorder=word_endian_order)
TypeError: fromCoils() got an unexpected keyword argument 'wordorder'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\ye\AppData\Local\Programs\Python\Python39\lib\site-packages\thingsboard_gateway\connectors\modbus\bytes_modbus_uplink_converter.py", line 65, in convert
    decoder = BinaryPayloadDecoder.fromCoils(coils, wordorder=word_endian_order)
TypeError: fromCoils() got an unexpected keyword argument 'wordorder'
""2023-09-10 22:19:29" - |DEBUG| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 93 - {'deviceName': '设备2', 'deviceType': 'default', 'telemetry': [], 'attributes': []}"
""2023-09-10 22:19:34" - |DEBUG| - [modbus_connector.py] - modbus_connector - __process_slaves - 302 - Device connector type: tcp"
""2023-09-10 22:19:34" - |DEBUG| - [modbus_connector.py] - modbus_connector - __function_to_device - 515 - With result ReadCoilsResponse(8)"
""2023-09-10 22:19:34" - |DEBUG| - [modbus_connector.py] - modbus_connector - __process_slaves - 338 - Checking timeseries for device 设备2"
""2023-09-10 22:19:34" - |DEBUG| - [modbus_connector.py] - modbus_connector - __process_slaves - 339 - Device response: "
""2023-09-10 22:19:34" - |ERROR| - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - convert - 92 - fromCoils() got an unexpected keyword argument 'wordorder'"
Traceback (most recent call last):
  File "C:\Users\ye\AppData\Local\Programs\Python\Python39\lib\site-packages\thingsboard_gateway\connectors\modbus\bytes_modbus_uplink_converter.py", line 63, in convert
    decoder = BinaryPayloadDecoder.fromCoils(coils, byteorder=endian_order, wordorder=word_endian_order)
TypeError: fromCoils() got an unexpected keyword argument 'wordorder'
![image](https://github.com/thingsboard/thingsboard-gateway/assets/19941827/cb6fdef1-68f7-429a-bd73-4df7710b8f43)
![image](https://github.com/thingsboard/thingsboard-gateway/assets/19941827/8244b777-dc5a-4d32-ac62-217f52c0cfe5)

'modbus.json'
{
  "master": {
    "slaves": [
      {
        "host": "127.0.0.1",
        "port": 502,
        "type": "tcp",
        "method": "socket",
        "timeout": 35,
        "byteOrder": "LITTLE",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 1,
        "deviceName": "设备2",
        "deviceType": "default",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 300000,
        "attributes": [],
        "timeseries": [
          {
            "tag": "CH1",
            "type": "8int",
            "functionCode": 1,
            "objectsCount": 1,
            "address": 0
          }
        ]
      }
    ]
  },
  "slave": {}
}

**Versions (please complete the following information):**
 - OS: windwos 11
 - Thingsboard IoT Gateway version [e.g. 3.3]
 - Python version[e.g. 3.9]

samson0v, 2023-09-11:
Hi @struggle3, thanks for your interest in ThignsBoard IoT Gateway!
This bug was fixed.
You can use the version from the master branch or wait for release.

MaryLee7979, 2023-09-25:
@struggle3 我曾經遇過這問題, 大概是他們在9/12改版的時候有出現該情況, 後來我手頭有8/11的connectors/modbus的文件覆蓋。即解決該問題。

[modbus.zip](https://github.com/thingsboard/thingsboard-gateway/files/12713071/modbus.zip)

struggle3, 2023-09-25:
好的，谢谢，我已经解决了。

closed 2023-09-29 by samson0v