thingsboard/thingsboard-gateway #1181: [HELP]

MaryLee7979, 2023-09-21:
Question 1. Gateway's tb_gateway.yml has been changed to json format. Is there a more detailed description of the "connectors" parameter? The official website information is still in yml format.

Question 2. Trying to set tb_gateway.json and start tb_gateway.py through python, but the following error occurs and cannot be run.

**Configuration** (Attach your configuration file)
[tb_gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/12681672/tb_gateway.txt)

[image]

**Error traceback (If it was raised):**
[image]

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version [3.3]
 - Python version:3.9.5

imbeacon, 2023-09-21:
Hi, @MaryLee7979 

Yes, starting from the version 3.4 we decided to use JSON instead of yaml to decrease amount of required libraries and unify configuration files. You should have tb_gateway.json in the configuration folder with your configurations. If no - please let us know about this.

imbeacon, 2023-09-21:
Documentation is in progress and it will updated as soon as possible.

MaryLee7979, 2023-09-25:
> Hi, @MaryLee7979
> 
> Yes, starting from the version 3.4 we decided to use JSON instead of yaml to decrease amount of required libraries and unify configuration files. You should have tb_gateway.json in the configuration folder with your configurations. If no - please let us know about this.

iot gateway v3.4 the package contains json format files,Then I will run it using v3.3 for now and wait for your completed operating instructions for 3.4.
Thanks you.

mrangics, 2023-10-03:
Would really helpful to provide a sample configuration file with the new JSON format including the OOTB connectors, even here, just to post it.

devaskim, 2023-10-11:
@mrangics Did you mean [this file from the repo](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/config/tb_gateway.json) ?

mrangics, 2023-10-11:
> @mrangics Did you mean [this file from the repo](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/config/tb_gateway.json) ?
> 
> ```
> ...
>  "connectors": [
>     {
>     // the same fileds as in yaml but with json formatting
>     }
> ]
> ...
> ```

Hi,
yes, but this is not complete. A complete JSON is necessary to show how to define the connectors!
Thanks,
Máté

devaskim, 2023-10-11:
@mrangics Agree, it is not complete and that might be a problem for new users. For others, it is just YAML to JSON conversion of existing configuration and there is a lot online converters

devaskim, 2023-10-11:
In general, it is sad that there is no caution at least [on the release page](https://github.com/thingsboard/thingsboard-gateway/releases) about migrating to JSON format.

samson0v, 2023-10-11:
Hi @devaskim and @MaryLee7979, thanks for your interest in ThingsBoard IoT Gateway!
Documentation [PR](https://github.com/thingsboard/thingsboard.github.io/pull/1166) created, waiting for merge.

devaskim, 2023-10-13:
@samson0v Great! Now I am calm for humanity ;)

samson0v, 2023-10-30:
Updated docs:
https://thingsboard.io/docs/iot-gateway/guides/how-to-enable-remote-configuration/
https://thingsboard.io/docs/iot-gateway/getting-started/

closed 2023-10-30 by samson0v