thingsboard/thingsboard-gateway #1182: Thingsboard does net receive Data via TCP Socket [HELP]

BAslan1988, 2023-09-22:
**Describe the issue**

I am trying to send data to the Thingsboard gateway. I have set up the Socket Connector as the only connector.  After setting up the TB gateway, it was able to connect successfully to the existing Thingsboard. To test the socket connection, I sent a simple netcat command.
`echo "1234" | nc localhost 50001`
 However, no bytes are registered in thingboard . No logs are written in the connector logs either. Please take a look at the configuration and tell me if everything is set up correctly.
**Configuration** (Attach your configuration file)
**socket.json**
```{
  "name": "TCP_Connector",
  "type": "TCP",
  "address": "127.0.0.1",
  "port": 50000,
  "bufferSize": 1024,
  "loglevel": "DEBUG",
  "devices": [
    {
      "address": "127.0.0.1:50001",
      "deviceName": "Device_Example",
      "accessToken": "Device_example",
      "deviceType": "default",
      "encoding": "utf-8",
      "telemetry": [
        {
          "key": "temperature",
          "byteFrom": 0,
          "byteTo": -1
        }
      ]
    }
  ]
}
```

**tb_gateway.json**

```{
  "thingsboard": {
    "host": "127.0.0.1",
    "port": 1883,
    "remoteShell": true,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 60
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 60,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 10,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "MY_ACCESS_TOKEN"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 10000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
    {
      "loglevel": "DEBUG",
      "name": "TCP_Connector",
      "type": "socket",
      "configuration": "socket.json"
    }
  ]
}
```

**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
I need help with the Socket Connector to establish a Data Stream via TCP.

**Error traceback (If it was raised):**
**Logs are empty**
Except on Start up:

```
2023-09-22 07:12:55 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 196 - Gateway starting...
2023-09-22 07:12:55 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 201 - ThingsBoard IoT gateway version: 3.4
2023-09-22 07:12:55 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 184 - Subscribed to *|* with id 1 for device *
2023-09-22 07:12:55 - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - init_remote_shell - 388 - Remote shell is enabled. Please be carefully with this feature.
2023-09-22 07:12:55 - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import SocketConnector from /thingsboard_gateway/connectors/socket.
2023-09-22 07:12:55 - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import BytesSocketUplinkConverter from /thingsboard_gateway/connectors/socket.
2023-09-22 07:12:55 - |INFO| - [socket_connector.py] - socket_connector - run - 156 - TCP socket is up
2023-09-22 07:12:55 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - __init__ - 54 - Remote Configurator started
```

**Versions (please complete the following information):**
 - OS: [ **RED Rocky Linux**]
 - Thingsboard IoT Gateway version [3.4]
 - Python version[3.9]

BAslan1988, 2023-09-26:
How has the message payload has to look like? I can't find any information about this in the Socket Connector Guide.

samson0v, 2023-09-29:
Hi @BAslan1988, thanks for your interest in ThingsBoard IoT Gateway!
I created a PR to fix this issue.
But, also, you have to add the `logLevel` parameter to the top of the connector config file, instead of adding it to the tb_gateway.json connector section.
[image]

samson0v, 2023-09-29:
@BAslan1988, so for now, please, use the version from the master branch.

closed 2023-10-10 by samson0v