thingsboard/thingsboard-gateway #1184: [HELP] after upgrading I get "MQTT Broker Connector connection FAIL with error 134 Bad User Name or Password!"

keestux, 2023-09-30:
**Describe the issue**
After upgrading to version 3.3 I'm seeing the following message in my syslog. After upgrading to version 3.4 the messages are still produced.
```
Sep 30 19:04:30 my.example.com python3[70652]: ""2023-09-30 19:04:30" - |ERROR| - [mqtt_connector.py] - mqtt_connector - _on_connect - 441 - MQTT Broker Connector connection FAIL with error 134 Bad User Name or Password!"
```

Is this maybe a new error message for a missing MQTT configuration?

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
```
root@test:~# cat /etc/thingsboard-gateway/config/tb_gateway.yaml | grep -v '^#'
thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statistics:
    enable: true
    statsSendPeriodInSeconds: 3600
    configuration: statistics.json
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: PUT_YOUR_GW_ACCESS_TOKEN_HERE
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
grpc:
  enabled: false
  serverPort: 9595
  keepaliveTimeMs: 10000
  keepaliveTimeoutMs: 5000
  keepalivePermitWithoutCalls: true
  maxPingsWithoutData: 0
  minTimeBetweenPingsMs: 10000
  minPingIntervalWithoutDataMs: 5000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
```

**Connector name (If you need help with some connector/converter):**
[MQTT Connector]

syslog messages
```
Sep 30 19:12:36 my.example.com python3[70652]: ""2023-09-30 19:12:36" - |ERROR| - [mqtt_connector.py] - mqtt_connector - _on_connect - 441 - MQTT Broker Connector connection FAIL with error 134 Bad User Name or Password!"
```

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04
 - Thingsboard IoT Gateway version: 3.4
 - Python version: 3.10.12

LukaszDuss, 2023-10-09:
`
accessToken: PUT_YOUR_GW_ACCESS_TOKEN_HERE
`
did you replace PUT_YOUR_GW_ACCESS_TOKEN_HERE with actual token?

keestux, 2023-10-09:
@LukaszDuss no I haven't changed it. I guess I should have.
The point is there were never any of these MQTT errors before, until I upgraded to version 3.3. Can you confirm that the misconfiguration was improved since version 3.3?

LukaszDuss, 2023-10-10:
My guess would be that the update replaced Your tb_gateway.yaml file with default one. 
So you lost token(password) that enabled GW to connect to mqtt broker.

keestux, 2023-10-10:
@LukaszDuss if that would be true, then you need to change the installation procedure. Such vital part of the configuration should never be overwritten during an upgrade.

However, I'm not sure if that is what really happened. I'm guessing this token was never filled in. I'm also guessing that the newer versions complain more about a missing token then the older versions. And that that is the reason why I see this message. Could that be it?

Can you tell me how to make a new token? And is this the only place where I fill in that token?

keestux, 2023-10-10:
In the mean time I found the access token. Indeed the file is simply replaced during an upgrade.

Still, the error remains. It keeps giving
```
Oct 10 18:57:14 my.example.com python3[149354]: ""2023-10-10 18:57:14" - |ERROR| - [mqtt_connector.py] - mqtt_connector - _on_connect - 441 - MQTT Broker Connector connection FAIL with error 134 Bad User Name or Password!"
```

keestux, 2023-10-12:
OK. I came to the conclusion that a package upgrade (Ubuntu .deb) will completely destroy an existing configuration. This issue can be closed.

closed 2023-10-12 by keestux