thingsboard/thingsboard-gateway #1185: [HELP] Modbus Connector is unable to start ()

mrangics, 2023-10-03:
**Describe the issue**
Somehow the Modbus connector is not able to start. Most probably the configuration file is not correct?
Error is:
|ERROR| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 609 - 'wordOrder'"

**Configuration**
[tb_gateway.yaml.txt](https://github.com/thingsboard/thingsboard-gateway/files/12795782/tb_gateway.yaml.txt)
[modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/12795783/modbus.json.txt)

**Connector name (If you need help with some connector/converter):**
Modbus Connector

**Error traceback (If it was raised):**
```
2023-10-03 20:27:00 Requirement already satisfied: mmh3 in /root/.local/lib/python3.10/site-packages (4.0.1)
2023-10-03 20:27:01 WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 148 - Gateway starting..."
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 153 - ThingsBoard IoT gateway version: 3.2"
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 184 - Subscribed to *|* with id 1 for device *"
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import GrpcModbusConnector from /thingsboard_gateway/grpc_connectors/modbus."
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import ModbusConnector from /thingsboard_gateway/connectors/modbus."
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |ERROR| - [tb_gateway_service.py] - tb_gateway_service - _connect_with_connectors - 609 - 'wordOrder'"
2023-10-03 20:27:01 Traceback (most recent call last):
2023-10-03 20:27:01   File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 599, in _connect_with_connectors
2023-10-03 20:27:01     connector = self._implemented_connectors[connector_type](self,
2023-10-03 20:27:01   File "/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 134, in __init__
2023-10-03 20:27:01     self.__load_slaves()
2023-10-03 20:27:01   File "/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 210, in __load_slaves
2023-10-03 20:27:01     self.__slaves = [
2023-10-03 20:27:01   File "/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 211, in <listcomp>
2023-10-03 20:27:01     Slave(**{**device, 'connector': self, 'gateway': self.__gateway, 'callback': ModbusConnector.callback}) for
2023-10-03 20:27:01   File "/thingsboard_gateway/connectors/modbus/slave.py", line 43, in __init__
2023-10-03 20:27:01     'wordOrder': kwargs['wordOrder'],
2023-10-03 20:27:01 KeyError: 'wordOrder'
2023-10-03 20:27:01 ""2023-10-03 18:27:01" - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 252 - Gateway started."
2023-10-03 20:27:02 ""2023-10-03 18:27:02" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - 3.4"
2023-10-03 20:27:02 ""2023-10-03 18:27:02" - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 - 
2023-10-03 20:27:02 
2023-10-03 20:27:02 [===UPDATE===]
2023-10-03 20:27:02 
2023-10-03 20:27:02  New version 3.4 is available! 
2023-10-03 20:27:02 
2023-10-03 20:27:02 [===UPDATE===]
2023-10-03 20:27:02 "
```

**Versions (please complete the following information):**
 - OS: Host OS Ubuntu 22.04
 - Thingsboard IoT Gateway version 3.2.0, Docker
 - Python version 3.9
 - Pymodbus version 3.0 (downgraded to avoid error)

LukaszDuss, 2023-10-09:
check  MASTER/SLAVE config file instead of SERVER
[thingsboard_gateway/config/modbus.json ](https://github.com/thingsboard/thingsboard-gateway/blob/1f2b4d662f6613eb519a66aa290f28eed0ec4bc5/thingsboard_gateway/config/modbus.json)

mrangics, 2023-10-09:
OK, thank you, let me try and give you a feedback.

closed 2023-10-10 by samson0v