thingsboard/thingsboard-gateway #1189: [BUG] Using "bit" in modbus connector doesn't work anymore

EinarArnason, 2023-10-06:
**Describe the bug**
Defining "bit" in timeseries throws an error í 3.4

`"timeseries": [
          {
            "byteOrder": "BIG",
            "tag": "relay",
            "type": "bits",
            "functionCode": 3,
            "objectsCount": 1,
            "address": 1,
            "bit": 8
          }
]`

**Connector name (If bug in the some connector):**
Modbus connector

**Error traceback (If available):**
```
thingsboard-gateway[3539]: Traceback (most recent call last):
thingsboard-gateway[3539]:   File "/home/thingsboard/.local/pipx/venvs/thingsboard-gateway/lib/python3.11/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 82, in convert
thingsboard-gateway[3539]:     decoded_data = self.decode_from_registers(decoder, configuration)
thingsboard-gateway[3539]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
thingsboard-gateway[3539]:   File "/home/thingsboard/.local/pipx/venvs/thingsboard-gateway/lib/python3.11/site-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 167, in decode_from_registers
thingsboard-gateway[3539]:     result_data = int(decoded[configuration['bit']])
thingsboard-gateway[3539]:                       ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
thingsboard-gateway[3539]: IndexError: list index out of range
thingsboard-gateway[3539]: 12:55:56 ERROR tb_logger:97 list index out of range
```

**Versions (please complete the following information):**
 - OS: Arch linux
 - Thingsboard IoT Gateway version 3.4
 - Python version 3.11

EinarArnason, 2023-10-06:
No matter what number I put in there, still fails.

imbeacon, 2023-10-06:
Could you set bits to 16 instead of 1. Because it reads set amount of bits from address. Bit is optional and it is just an index of target bit.

EinarArnason, 2023-10-06:
Worked. Setting objectsCount to 16 fixed the issue.

closed 2023-10-06 by EinarArnason