thingsboard/thingsboard-gateway #1195: [BUG] MQTT connector does not re-connect after error.

fjpanag, 2023-10-14:
**Describe the bug**
It occurred to me that there was an error with the MQTT broker that the Gateway was trying to connect with.

This is its output:

```
Exception in thread TEST:
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 298, in run
    self.__connect()
  File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 305, in __connect
    self._client.connect(self.__broker['host'],
  File "/usr/local/lib/python3.11/site-packages/paho/mqtt/client.py", line 915, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/paho/mqtt/client.py", line 1086, in reconnect
    sock.do_handshake()
  File "/usr/local/lib/python3.11/ssl.py", line 1379, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: [SSL: UNEXPECTED_EOF_WHILE_READING] EOF occurred in violation of protocol (_ssl.c:1006)
```

and later in the log:

```
2023-10-14 14:28:29 - |ERROR| - [tb_logger.py] - tb_logger - exception - 97 - [Errno 104] Connection reset by peer
Traceback (most recent call last):
  File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 286, in run
    self.__connect()
  File "/thingsboard_gateway/connectors/mqtt/mqtt_connector.py", line 305, in __connect
    self._client.connect(self.__broker['host'],
  File "/usr/local/lib/python3.11/site-packages/paho/mqtt/client.py", line 915, in connect
    return self.reconnect()
           ^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/paho/mqtt/client.py", line 1086, in reconnect
    sock.do_handshake()
  File "/usr/local/lib/python3.11/ssl.py", line 1379, in do_handshake
    self._sslobj.do_handshake()
ConnectionResetError: [Errno 104] Connection reset by peer
```

This is not an issue, as I know that the broker was just starting up, and was not fully ready.

However, the problem is that **Gateway does not try to reconnect at a later time**!  
It fails in this state, and it cannot recover.

**A manual restart of the Gateway is needed for it to recover.**

I would expect, either Gateway to retry the connection after a few seconds, or at least exit (with non-zero status), so my system can catch the issue and perform a restart of the service.

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
*See above.*

**Versions (please complete the following information):**  
 - OS: Docker deployment on a Rocky 9.
 - Thingsboard IoT Gateway version: *(not stated in logs!)*
 - Python version: 3.11.6

imbeacon, 2024-01-26:
Hi @fjpanag,

Thank you for your interest in Thingsboard IoT Gateway. You are right and bug was confirmed, for now - it is fixes, in the master branch, please use it or wait for the nearest release.

closed 2024-01-26 by imbeacon

volpima, 2024-04-17:
Hello @imbeacon, I had a similar issue with version 3.4.6:

```
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2024-04-16 17:32:48 CEST; 15h ago
   Main PID: 944 (python3)
      Tasks: 17 (limit: 961)
     Memory: 57.6M
        CPU: 16h 44min 9.207s
     CGroup: /system.slice/thingsboard-gateway.service
             └─944 /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Apr 16 19:08:09 ubuntu python3[944]:     rc = self.loop_write()
Apr 16 19:08:09 ubuntu python3[944]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1590, in loop_write
Apr 16 19:08:09 ubuntu python3[944]:     rc = self._packet_write()
Apr 16 19:08:09 ubuntu python3[944]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2510, in _packet_write
Apr 16 19:08:09 ubuntu python3[944]:     write_length = self._sock_send(
Apr 16 19:08:09 ubuntu python3[944]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/paho/mqtt/client.py", line 650, in _sock_send
Apr 16 19:08:09 ubuntu python3[944]:     return self._sock.send(buf)
Apr 16 19:08:09 ubuntu python3[944]:   File "/usr/lib/python3.10/ssl.py", line 1235, in send
Apr 16 19:08:09 ubuntu python3[944]:     return self._sslobj.write(data)
Apr 16 19:08:09 ubuntu python3[944]: ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:2426)
```
Could you please investigate?
It happens after some hour, needed restart then it stops again after some time.
Thankyou.

volpima, 2024-05-03:
Hello! 
@imbeacon any news about this?