thingsboard/thingsboard-gateway #1204: [HELP] Can't make the new device to be created in the TB environment after establishing gateway connection with the OPC-UA server simulation

IhorYednak, 2023-10-26:
So I have a gateway setup on ubuntu as well as a the OPC-UA server simulator, and I follow this guide - https://thingsboard.io/docs/iot-gateway/config/opc-ua/?opcuaIdentityConfig=anonymous in order to setup this connector and create a new device automatically in my environment. But something seems to be wrong as the device can’t be created. Attaching a code of the tb-gateway.yaml config file and opcua.json files and a screenshot of the OPC-UA server status page. Could you help me with this?

tb_gateway.yaml

```
{
  "thingsboard": {
    "host": "sandbox.produmize.com",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 60
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 60,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 10,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "WlQv3KIrx3vqkbGKFx7E"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 10000
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
#    {
#      "type": "mqtt",
#      "name": "MQTT Broker Connector",
#      "configuration": "mqtt.json"
#    },
#    {
#      "type": "modbus",
#      "name": "Modbus Connector",
#      "configuration": "modbus.json"
#    },
#    {
#      "type": "modbus",
#      "name": "Modbus Serial Connector",
#      "configuration": "modbus_serial.json"
#    },
    {
      "type": "opcua",
      "name": "OPC-UA Connector",
      "configuration": "opcua.json"
    }
  ]
}
```
 

opcua.json

```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "opc.tcp://ihor-VirtualBox:53530/OPCUA/SimulationServer",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions": false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "username": "user",
      "password": "5Tr0nG?@$sW0rD"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Simulation",
        "deviceNamePattern": "DeviceOPCUA",
        "attributes": [
        ],
        "timeseries": [
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [
              2,
              4
            ]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Simulation\\.serialNumber"
          }
        ]
      }
    ]
  }
}
![Screenshot_4](https://github.com/thingsboard/thingsboard-gateway/assets/54800500/c1417650-f810-4f18-bd3d-f898f50adc4f)

samson0v, 2023-10-31:
Hi @IhorYednak, thanks for your interest in ThingsBoard IoT Gateway!
You have to provide attributes or time series in the opc-ua config file. Gateway doesn't send data without a payload, so the device will not be created.

IhorYednak, 2023-11-01:
Hi @samson0v , thank you for your response, could you please also let me know about these paths listed in opcua.json, I'm running an OPC-UA simulator and not sure about what files should be listed under these paths. And also I keep getting this error 
[image] after I restart the gateway and check the status of it

IhorYednak, 2023-11-08:
@samson0v , are there any suggestions?

closed 2024-04-23 by IhorYednak