thingsboard/thingsboard-gateway #1215: [BUG] TBGW 3.4.2 BACnet [Errno 98] Address already in use

HarvUK, 2023-11-05:
**Describe the bug**
Following PR [#1213](https://github.com/thingsboard/thingsboard-gateway/pull/1213)
I have copied the 2 amended .py files into the container (a completely new copy of the container).

However, with a very simple config (done using remote configuration- great tool!), I get the error log below.

This is the only connector active on this device + container, and I've tried restarting it too, I'm stumped as to what to do - 172.17.0.2 is the IP address of the container, so that's correct.

No data ever arrives in our thignsboard instance from this connector on this container.

**Connector name (If bug in the some connector):**
BACnet

**Error traceback (If available):**
2023-11-05 21:18:06 2023-11-05 21:18:06 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_active_connectors_update - 340 - Processing active connectors configuration update...
2023-11-05 21:18:07 2023-11-05 21:18:07 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - process_config_request - 176 - Got config update request: {'bacnettest': {'name': 'bacnettest', 'type': 'bacnet', 'logLevel': 'DEBUG', 'configuration': 'bacnettest.json', 'configurationJson': {'general': {'objectName': 'TB_gateway', 'address': '172.17.0.2:47808', 'objectIdentifier': 602, 'maxApduLengthAccepted': 1476, 'segmentationSupported': 'segmentedBoth', 'vendorIdentifier': 15}, 'devices': [{'deviceName': 'ADevice', 'deviceType': 'default', 'address': '192.168.21.242:47808', 'pollPeriod': 300000, 'timeseries': [{'key': 'Buffer Top Temp', 'objectId': 'analogInput:0', 'propertyId': 'presentValue'}]}]}, 'ts': 1699219088284}}
2023-11-05 21:18:07 2023-11-05 21:18:07 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 380 - Processing connectors configuration update...
2023-11-05 21:18:07 2023-11-05 21:18:07 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - create_configuration_file_backup - 595 - Backup file created for configuration file bacnettest.json in //thingsboard_gateway/config/backup/bacnettest.json_backup_1699219087
2023-11-05 21:18:07 2023-11-05 21:18:07 - |ERROR| - [tb_logger.py] - tb_logger - exception - 97 - [Errno 98] Address already in use
2023-11-05 21:18:07 Traceback (most recent call last):
2023-11-05 21:18:07   File "/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 42, in __init__
2023-11-05 21:18:07     super().__init__(self.__device, self.__config["general"]["address"])
2023-11-05 21:18:07   File "/root/.local/lib/python3.11/site-packages/bacpypes/app.py", line 535, in __init__
2023-11-05 21:18:07     self.mux = UDPMultiplexer(self.localAddress)
2023-11-05 21:18:07                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-11-05 21:18:07   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 96, in __init__
2023-11-05 21:18:07     self.directPort = UDPDirector(self.addrTuple)
2023-11-05 21:18:07                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-11-05 21:18:07   File "/root/.local/lib/python3.11/site-packages/bacpypes/udp.py", line 155, in __init__
2023-11-05 21:18:07     self.bind(address)
2023-11-05 21:18:07   File "/usr/local/lib/python3.11/asyncore.py", line 331, in bind
2023-11-05 21:18:07     return self.socket.bind(addr)
2023-11-05 21:18:07            ^^^^^^^^^^^^^^^^^^^^^^
2023-11-05 21:18:07 OSError: [Errno 98] Address already in use
2023-11-05 21:18:08 2023-11-05 21:18:08 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 205 - WhoIsRequest has been sent.
2023-11-05 21:18:08 2023-11-05 21:18:08 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - run - 78 - WhoIsRequest has been sent.
2023-11-05 21:18:08 2023-11-05 21:18:08 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 205 - WhoIsRequest has been sent.

'deviceName'
Traceback (most recent call last):
2023-11-05 21:18:07 Traceback (most recent call last):
2023-11-05 21:18:07   File "/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 42, in __init__
2023-11-05 21:18:07     super().__init__(self.__device, self.__config["general"]["address"])
2023-11-05 21:18:07   File "/root/.local/lib/python3.11/site-packages/bacpypes/app.py", line 535, in __init__
2023-11-05 21:18:07     self.mux = UDPMultiplexer(self.localAddress)
2023-11-05 21:18:07                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-11-05 21:18:07   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 96, in __init__
2023-11-05 21:18:07     self.directPort = UDPDirector(self.addrTuple)
2023-11-05 21:18:07                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-11-05 21:18:07   File "/root/.local/lib/python3.11/site-packages/bacpypes/udp.py", line 155, in __init__
2023-11-05 21:18:07     self.bind(address)
2023-11-05 21:18:07   File "/usr/local/lib/python3.11/asyncore.py", line 331, in bind
2023-11-05 21:18:07     return self.socket.bind(addr)

**Versions (please complete the following information):**
OS: Windows 11, Docker Desktop 4.24.2
Thingsboard IoT Gateway version [e.g. 2.0]
3.4.2
Python version[e.g. 3.7]
3.11.6

HarvUK, 2023-11-05:
Update,

1. Setting the general address to 0.0.0.0:47809 as I saw in an earlier closed [bug](https://github.com/thingsboard/thingsboard-gateway/issues/668) gives this error trace - this may be a different issue now?

2. Rebooting the container goes back to the original error above ([Errno 98] Address already in use), so it may be that the connector is not releasing the address when it has finished its communication?

2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - process_config_request - 176 - Got config update request: {'bacnettest': {'name': 'bacnettest', 'type': 'bacnet', 'logLevel': 'DEBUG', 'configuration': 'bacnettest.json', 'configurationJson': {'general': {'objectName': 'TB_gateway', 'address': '0.0.0.0:47809', 'objectIdentifier': 602, 'maxApduLengthAccepted': 1476, 'segmentationSupported': 'segmentedBoth', 'vendorIdentifier': 15}, 'devices': [{'deviceName': 'ADevice', 'deviceType': 'default', 'address': '192.168.21.242:47808', 'pollPeriod': 300000, 'timeseries': [{'key': 'Buffer Top Temp', 'objectId': 'analogInput:0', 'propertyId': 'presentValue'}]}]}, 'ts': 1699220009668}}
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 380 - Processing connectors configuration update...
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - create_configuration_file_backup - 595 - Backup file created for configuration file bacnettest.json in //thingsboard_gateway/config/backup/bacnettest.json_backup_1699220008
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 205 - WhoIsRequest has been sent.
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - run - 78 - WhoIsRequest has been sent.
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [bacnet_connector.py] - bacnet_connector - scan_network - 205 - WhoIsRequest has been sent.
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - indication - 68 - Received IAmRequest from device with ID: 1 and address 192.168.21.242:47808
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - indication - 73 - 192.168.21.242
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - indication - 68 - Received IAmRequest from device with ID: 1 and address 192.168.21.242:47808
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - indication - 73 - 192.168.21.242
2023-11-05 21:33:28 2023-11-05 21:33:28 - |DEBUG| - [tb_gateway_bacnet_application.py] - tb_gateway_bacnet_application - __iam_cb - 164 - Received IAm Response: <bacpypes.apdu.ReadPropertyACK(12,1) instance at 0x7f1e8e18d2d0>
2023-11-05 21:33:28 2023-11-05 21:33:28 - |ERROR| - [core.py] - core - run - 174 - an error has occurred: BACnetUplinkConverter.__init__() missing 1 required positional argument: 'logger'
2023-11-05 21:33:28 Traceback (most recent call last):
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/core.py", line 165, in run
2023-11-05 21:33:28     fn(*args, **kwargs)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/udp.py", line 301, in _response
2023-11-05 21:33:28     peer.response(pdu)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/udp.py", line 79, in response
2023-11-05 21:33:28     self.director.response(pdu)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
2023-11-05 21:33:28     self.serverPeer.confirmation(*args, **kwargs)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 41, in confirmation
2023-11-05 21:33:28     self.multiplexer.confirmation(self, pdu)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 179, in confirmation
2023-11-05 21:33:28     self.annexJ.response(PDU(pdu, source=src, destination=dest))
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
2023-11-05 21:33:28     self.serverPeer.confirmation(*args, **kwargs)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 312, in confirmation
2023-11-05 21:33:28     self.response(rpdu)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
2023-11-05 21:33:28     self.serverPeer.confirmation(*args, **kwargs)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/bvllservice.py", line 402, in confirmation
2023-11-05 21:33:28     self.response(xpdu)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/comm.py", line 320, in response
2023-11-05 21:33:28     self.serverPeer.confirmation(*args, **kwargs)
2023-11-05 21:33:28   File "/root/.local/lib/python3.11/site-packages/bacpypes/netservice.py", line 213, in confirmation
2023-11-05 21:33:28     self.adapterSAP.process_npdu(self, npdu)
[... cut ...]
2023-11-05 21:33:28     fn(self, *args, **kwargs)
2023-11-05 21:33:28   File "/thingsboard_gateway/connectors/bacnet/bacnet_utilities/tb_gateway_bacnet_application.py", line 167, in __iam_cb
2023-11-05 21:33:28     value = self.__connector.default_converters["uplink_converter"]("{}").convert(None, apdu)
2023-11-05 21:33:28             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2023-11-05 21:33:28 TypeError: BACnetUplinkConverter.__init__() missing 1 required positional argument: 'logger'

HarvUK, 2023-11-07:
I have resinstalled and reconfigured a couple more times, and it does seem to be that the issue lies in 

2023-11-05 21:33:28 2023-11-05 21:33:28 - |ERROR| - [core.py] - core - run - 174 - an error has occurred: BACnetUplinkConverter.init() missing 1 required positional argument: 'logger'

Could this relate to the previous commit fixing a logging issue?

samson0v, 2023-11-08:
Hi @HarvUK, thanks for your interest in ThingsBoard IoT Gateway!
The bug was fixed via #1219 PR.

closed 2023-11-08 by samson0v