thingsboard/thingsboard-gateway #1217: [HELP] Offline instalation

LukaszDuss, 2023-11-07:
**Describe the issue**
I can't install gateway using docker on offline vm 

I had downloaded docker image but container still needs connection to internet to update linux and python packages. 

Does anyone have some kind of workaround?

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 22.04]
 - Thingsboard IoT Gateway version [e.g. 3.4.2]

samy4sam, 2023-11-08:
I'm struggling with exactly the same problem.
[image]

The connectivity tests for http:
``curl -v -X POST http://<LOCAL_IP>:8080/api/v1/<TOKE>/telemetry --header Content-Type:application/json --data "{temperature:25}"``
and mqtt:
``mosquitto_pub -d -q 1 -h <LOCAL_IP> -p 1883 -t v1/devices/me/telemetry -u <TOKEN> -m "{temperature:25}"``
work without any problems.

As a temporary fix, I installed Thingsboard Gateway with pip without Docker: [Installation Guide](https://thingsboard.io/docs/iot-gateway/install/pip-installation/)

imbeacon, 2024-06-19:
Hi,

The first installation and configuration require an internet connection to download the required packages, unfortunately, it cannot be changed.

closed 2024-06-19 by imbeacon