thingsboard/thingsboard-gateway #1221: [BUG] Connector not working, Module not found

samy4sam, 2023-11-08:
**Describe the bug**
I get an error on the Thingsboard Gateway when I try to add a connector via the Gateway Dashboard. The corresponding module is not found. The error occurs with OPC-UA, REST and BLE. The MQTT connector works so far. I have not tested the others.

I have the same bug with Docker running on Windows and Ubuntu.

[image]

**Connector name (If bug in the some connector):**
[OPC-UA Connector]
[REST Connector]
[BLE Connector]

**Error traceback (If available):**
OPC-UA Connector:
```
2023-11-08 20:45:57 OPC-UA library not found
2023-11-08 20:45:58 Collecting opcua
2023-11-08 20:45:58   Downloading opcua-0.98.13.tar.gz (572 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 572.8/572.8 kB 2.9 MB/s eta 0:00:00
2023-11-08 20:45:59   Preparing metadata (setup.py) ... done
2023-11-08 20:45:59 Collecting lxml (from opcua)
2023-11-08 20:45:59   Obtaining dependency information for lxml from https://files.pythonhosted.org/packages/ed/62/ffc30348ae141f69f9f23b65ba769db7ca209856c9a9b3406279e0ea24de/lxml-4.9.3-cp311-cp311-manylinux_2_28_x86_64.whl.metadata
2023-11-08 20:45:59   Downloading lxml-4.9.3-cp311-cp311-manylinux_2_28_x86_64.whl.metadata (3.8 kB)
2023-11-08 20:45:59 Collecting python-dateutil (from opcua)
2023-11-08 20:45:59   Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 247.7/247.7 kB 3.8 MB/s eta 0:00:00
2023-11-08 20:45:59 Collecting pytz (from opcua)
2023-11-08 20:45:59   Obtaining dependency information for pytz from https://files.pythonhosted.org/packages/32/4d/aaf7eff5deb402fd9a24a1449a8119f00d74ae9c2efa79f8ef9994261fc2/pytz-2023.3.post1-py2.py3-none-any.whl.metadata
2023-11-08 20:45:59   Downloading pytz-2023.3.post1-py2.py3-none-any.whl.metadata (22 kB)
2023-11-08 20:46:00 Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.11/site-packages (from python-dateutil->opcua) (1.16.0)
2023-11-08 20:46:00 Downloading lxml-4.9.3-cp311-cp311-manylinux_2_28_x86_64.whl (7.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 7.9/7.9 MB 3.0 MB/s eta 0:00:00
2023-11-08 20:46:02 Downloading pytz-2023.3.post1-py2.py3-none-any.whl (502 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 502.5/502.5 kB 3.9 MB/s eta 0:00:00
2023-11-08 20:46:02 Building wheels for collected packages: opcua
2023-11-08 20:46:03   Building wheel for opcua (setup.py) ... done
2023-11-08 20:46:03   Created wheel for opcua: filename=opcua-0.98.13-py3-none-any.whl size=628133 sha256=2cf79d8ec99f67550b7e496e49b696645599e32e7d16fa55ad1e1acfd6a8cbda
2023-11-08 20:46:03   Stored in directory: /root/.cache/pip/wheels/b0/6b/50/59e5a97aa96b6a62814e40e12f987a10df4980754f5ed5641e
2023-11-08 20:46:03 Successfully built opcua
2023-11-08 20:46:03 Installing collected packages: pytz, python-dateutil, lxml, opcua
2023-11-08 20:46:06 Successfully installed lxml-4.9.3 opcua-0.98.13 python-dateutil-2.8.2 pytz-2023.3.post1
2023-11-08 20:46:06 WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv
2023-11-08 20:46:06 
2023-11-08 20:46:06 [notice] A new release of pip is available: 23.2.1 -> 23.3.1
2023-11-08 20:46:06 [notice] To update, run: pip install --upgrade pip
2023-11-08 20:46:06 2023-11-08 19:46:06 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'opcua'
2023-11-08 20:46:06 Traceback (most recent call last):
2023-11-08 20:46:06   File "/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 33, in <module>
2023-11-08 20:46:06     from opcua import Client, Node, ua
2023-11-08 20:46:06 ModuleNotFoundError: No module named 'opcua'
2023-11-08 20:46:06 
2023-11-08 20:46:06 During handling of the above exception, another exception occurred:
2023-11-08 20:46:06 
2023-11-08 20:46:06 Traceback (most recent call last):
2023-11-08 20:46:06   File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
2023-11-08 20:46:06     module_spec.loader.exec_module(module)
2023-11-08 20:46:06   File "<frozen importlib._bootstrap_external>", line 940, in exec_module
2023-11-08 20:46:06   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2023-11-08 20:46:06   File "/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 37, in <module>
2023-11-08 20:46:06     from opcua import Client, Node, ua
2023-11-08 20:46:06 ModuleNotFoundError: No module named 'opcua'
2023-11-08 20:46:06 OPC-UA library not found
2023-11-08 20:46:06 Requirement already satisfied: opcua in /root/.local/lib/python3.11/site-packages (0.98.13)
2023-11-08 20:46:07 Requirement already satisfied: lxml in /root/.local/lib/python3.11/site-packages (from opcua) (4.9.3)
2023-11-08 20:46:07 Requirement already satisfied: python-dateutil in /root/.local/lib/python3.11/site-packages (from opcua) (2.8.2)
2023-11-08 20:46:07 Requirement already satisfied: pytz in /root/.local/lib/python3.11/site-packages (from opcua) (2023.3.post1)
2023-11-08 20:46:07 Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.11/site-packages (from python-dateutil->opcua) (1.16.0)
```

REST Connector:
[... 41 lines cut ...]
2023-11-08 20:52:47 During handling of the above exception, another exception occurred:
2023-11-08 20:52:47 
2023-11-08 20:52:47 Traceback (most recent call last):
2023-11-08 20:52:47   File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
2023-11-08 20:52:47     module_spec.loader.exec_module(module)
2023-11-08 20:52:47   File "<frozen importlib._bootstrap_external>", line 940, in exec_module
2023-11-08 20:52:47   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
2023-11-08 20:52:47   File "/thingsboard_gateway/connectors/rest/rest_connector.py", line 47, in <module>
2023-11-08 20:52:47     from aiohttp import web, BasicAuth
2023-11-08 20:52:47 ModuleNotFoundError: No module named 'aiohttp'
2023-11-08 20:52:47 2023-11-08 19:52:47 - |WARNING| - [tb_gateway_service.py] - tb_gateway_service - _load_connectors - 732 - Connector implementation not found for REST
2023-11-08 20:55:42 OPC-UA library not found
2023-11-08 20:55:43 Requirement already satisfied: opcua in /root/.local/lib/python3.11/site-packages (0.98.13)
2023-11-08 20:55:44 Requirement already satisfied: lxml in /root/.local/lib/python3.11/site-packages (from opcua) (4.9.3)
2023-11-08 20:55:44 Requirement already satisfied: python-dateutil in /root/.local/lib/python3.11/site-packages (from opcua) (2.8.2)
2023-11-08 20:55:44 Requirement already satisfied: pytz in /root/.local/lib/python3.11/site-packages (from opcua) (2023.3.post1)
2023-11-08 20:55:44 Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.11/site-packages (from python-dateutil->opcua) (1.16.0)
```

**Versions (please complete the following information):**
 - OS: Windows 10
 - Thingsboard IoT Gateway version: 3.4.1 (Docker)
 
  - OS: Ubuntu 22.04
 - Thingsboard IoT Gateway version: 3.4.2 (Docker)

samy4sam, 2023-11-20:
Hi @imbeacon 

Thank you for the hint! This has solved my problem.

dmohns, 2023-11-20:
Hey @samy4sam 

I'm encountering the same issue. Could you explain a bit more how you solved it? I tried restarting the docker and disable/enable the connector via UI. But the issue persists.

dmohns, 2023-11-20:
Ok, that's odd.

ThingsBoard version: `3.6.1` using GCP k8s Monolith setup. Followed the steps from here: https://thingsboard.io/docs/user-guide/install/cluster/gcp-monolith-setup/
ThingsBoard Gateway: Local Docker with `latest` (3.4.2 at the time of writing)

The command I use to run the docker locally is

```sh
docker run -it --rm \
          -v ./tb-gateway/logs:/thingsboard_gateway/logs \
          -v ./tb-gateway/extensions:/thingsboard_gateway/extensions \
          -v ./tb-gateway/config:/thingsboard_gateway/config \
          --name tb-gateway \
          --platform linux/amd64 \
          -p 5000:5000 \
          thingsboard/tb-gateway
```

dmohns, 2023-11-20:
Ok, seem like the `--rm` is causing the issue. Removing that brings back the same behaviour as you describe, i.e. after restart it finds the installed library.

closed 2023-11-22 by samy4sam