thingsboard/thingsboard-gateway #1223: [BUG] TB Gateway 3.4.2 + Docker on RPI4

elgutierrez, 2023-11-09:
**Describe the bug**
I'm trying to run `tb-gateway` on RPI 4 using docker. I'm pulling the latest image (3.4.2), but when I run the `docker run` command, I get  `exec /bin/sh: exec format error`

**Error traceback (If available):**
```
sudo docker run -it   -v ~/.tb-gateway/logs:/thingsboard_gateway/logs   -v ~/.tb-gateway/extensions:/thingsboard_gateway/extensions   -v ~/.tb-gateway/config:/thingsboard_gateway/config   --name tb-gateway --restart unless-stopped   thingsboard/tb-gateway

exec /bin/sh: exec format error
```

**Versions (please complete the following information):**
 - OS: Raspbian
 - Thingsboard IoT Gateway: 3.4.2

Jonnik84, 2023-11-13:
Its because docker container is not made for Rpi. Pls guys, add Rpi container to official list.

samy4sam, 2023-11-14:
I have the same problem. If you take a look at the [Thingsboard Gateway Docker images](https://hub.docker.com/r/thingsboard/tb-gateway/tags), you will see that the ARM architecture is no longer supported since version 3.4: 
[image]

elgutierrez, 2023-11-23:
Thanks @simenkid! I had to install from the sources as the latest release still have Bacnet issues but it might be useful in the future

rajatksud, 2024-04-29:
this is because cryptography requires the rust compiler which can be installed by modifying the dockerfile 

`# Install Rust
RUN apt-get update && apt-get install -y curl && \
    curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh -s -- -y && \
    echo 'export PATH="$HOME/.cargo/bin:$PATH"' >> $HOME/.bashrc && \
    /bin/bash -c "source $HOME/.bashrc" && \
    rustup update`
    
 The tb_updater.py file has import version and this module cannot be found

rajatksud, 2024-04-29:
This is because rust compiler is not available within docker which is used by cryptography

imbeacon, 2024-06-20:
Hi,

We have created a separated tag for RPi - tb-gateway:3.5.1-rpi, please try to use it instead of latest. Unfortunately, at the moment we cannot combine them.

closed 2024-06-20 by imbeacon