thingsboard/thingsboard-gateway #1226: Help with OPC-UA Connector with IoT Gateway

insurf, 2023-11-10:
Hallo,
I am using a ThingsBoard PE instance and need assistance with the integration of an OPC-UA interface. Specifically, I would like to monitor the values from a Siemens S7 1500 with ThingsBoard.

I have successfully connected the ThingsBoard PE to the IoT Gateway and the Gateway to the S7 via OPC UA.
Using an OPC Client, I can see the paths to the individual nodes. (See screenshot "Path")
How do I have to configure the opc.json file to obtain the nodes status? For Example node in "pic2"

This would be the config (opcua.json) I am using on the Gateway:
{
"server": {
"name": "OPC-UA Default Server",
"url": "opc.tcp://192.168.0.1:4840",
"timeoutInMillis": 5000,
"scanPeriodInMillis": 5000,
"disableSubscriptions": false,
"subCheckPeriodInMillis": 100,
"showMap": false,
"security": "Basic128Rsa15",
"identity": {
"type": "anonymous"
},
"mapping": [
{
"devices": [
{
"deviceNodePattern": "Root\.Objects\.001=TMP+SS-100KF1_PLC\.Outputs",
"deviceNamePattern": "001=TMP+SS-100KF1_PLC",
"attributes": []
"timeseries": [
{
"key": "Dosierung Fe3Cl",
"path": "$.['ns=3;s=Dosierung Fe3Cl']"
}
]
}
]
}
]
}
}

The log shows:
opcua_connector - init - 64 - deviceNodePattern in mapping: {"devices": [{"deviceNodePattern": "Root\.Objects\.001=TMP+SS-100KF1_PLC", "deviceNamePattern": "001=TMP+SS-100KF1_PLC", "attributes": [], "timeseries": [{"key": "Dosierung Fe3Cl", "path": "$.['ns=3;s=Dosierung Fe3Cl']"}]}]} - not found, add property deviceNodePattern to processing this mapping"

Would kindly ask for your support here.
Thank you!

[image]
[image]

  OS: Ubuntu 20.04
    ThingsBoard: ThingsBoard PE

Gateway:
OS: Raspbian 11
Gateway: 3.4.2

samy4sam, 2023-11-20:
@insurf 

When you are using namespace identifier and node identifier to map the object, you need to configure your path like this:

``` json
{
"key": "Dosierung Fe3Cl",
"path": "${ns=3;s=Dosierung Fe3Cl}"
}
```

SimoneG97, 2023-12-07:
Just to add on @samy4sam comment, as far as I can see the quotes are included in the identifier (which also happened to me in various cases), so you should use:
```
{
    "key": "Dosierung Fe3Cl",
    "path": "${ns=3;s=\"Dosierung Fe3Cl\"}"
}
```

closed 2024-06-19 by imbeacon