thingsboard/thingsboard-gateway #1227: [BUG] Missing Python Modules in ThingsBoard Gateway (pymodbus and twisted)

llamandtea, 2023-11-13:
**Describe the bug**
The gateway has raised two ModuleNotFoundError exceptions related to missing modules 'pymodbus' and 'twisted'. The only enabled connector is the Modbus one.

**Connector name (If bug in the some connector):**
Modbus connector

**Error traceback (If available):**
```
2023-11-13 14:45:54 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'pymodbus'
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/connectors/modbus/slave.py", line 18, in <module>
    from pymodbus.constants import Defaults
ModuleNotFoundError: No module named 'pymodbus'
2023-11-13 14:46:02 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'twisted'
Traceback (most recent call last):
  File "/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 51, in <module>
    from twisted.internet import reactor
ModuleNotFoundError: No module named 'twisted'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 54, in <module>
    from twisted.internet import reactor
ModuleNotFoundError: No module named 'twisted'
2023-11-13 14:46:02 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'pymodbus'
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py", line 15, in <module>
    from pymodbus.constants import Endian
ModuleNotFoundError: No module named 'pymodbus'
2023-11-13 14:46:02 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'pymodbus'
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/connectors/modbus/bytes_modbus_uplink_converter.py", line 15, in <module>
    from pymodbus.constants import Endian
ModuleNotFoundError: No module named 'pymodbus'
2023-11-13 14:46:02 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'pymodbus'
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/grpc_connectors/modbus/slave.py", line 18, in <module>
    from pymodbus.constants import Defaults
ModuleNotFoundError: No module named 'pymodbus'
2023-11-13 14:46:02 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'pymodbus'
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/grpc_connectors/modbus/modbus_connector.py", line 21, in <module>
    from thingsboard_gateway.connectors.modbus.bytes_modbus_downlink_converter import BytesModbusDownlinkConverter
  File "/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py", line 15, in <module>
    from pymodbus.constants import Endian
ModuleNotFoundError: No module named 'pymodbus'
2023-11-13 14:46:02 - |ERROR| - [tb_loader.py] - tb_loader - import_module - 72 - No module named 'pymodbus'
Traceback (most recent call last):
  File "/thingsboard_gateway/tb_utility/tb_loader.py", line 65, in import_module
    module_spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/thingsboard_gateway/grpc_connectors/modbus/bytes_modbus_uplink_converter.py", line 15, in <module>
    from pymodbus.constants import Endian
ModuleNotFoundError: No module named 'pymodbus'
```

**Versions (please complete the following information):**
 - OS: NixOs 23.11
 - Thingsboard IoT Gateway version 3.4.2 (executed as a docker container)
 - Python version 3.11.6 (the version installed in the docker image)

HarvUK, 2023-11-13:
Hi,

I've tried to replicate this, but on my docker image, all the packages downloaded fine (including pymodbus and twisted).

This suggests to me there is a problem with connectivity to the internet from your docker container.

Does your container show as an active device on your thingsboard instance?

I don't have a modbus device to hand to connect to, so I can't do a full test.

Cheers,

Harv

llamandtea, 2023-11-14:
Hi, strangely enough when rebuilding the container this morning there was no problem in the installation of the dependencies, without editing my compose file in any way. The gateway works fine with my modbus device. Thanks for the reply and sorry for opening an issue about this,

closed 2023-11-22 by imbeacon