thingsboard/thingsboard-gateway #1232: [BUG] The did not docker, connection FAIL with error 5 not authorised because no judgment env

sanliakeman123, 2023-11-17:
**Describe the bug**
The credentials used are information from environment variables

**Error traceback (If available):**
```
2023-11-17 19:25:36 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 222 - connection FAIL with error 5 not authorised

```

**Versions (please complete the following information):**
 - OS: Windows 11 
 - Thingsboard IoT Gateway version [3.4.2]
 - Python version[3.10]

code location
thingsboard_gateway\gateway\tb_gateway_service.py
line 193
self.__modify_main_config()

imbeacon, 2023-11-28:
Hi @sanliakeman123,

Thank you for your interest in Thingsboard IoT Gateway.
You are right, environment variables have more priority than configuration parameters. If you don't pass the variable, the gateway will not use it and it will use values from the configuration file.

closed 2024-01-26 by imbeacon

DuXiaoChuang, 2024-01-29:
how to resolve this problem??
[image]

tangchengxiang666, 2024-05-29:
> 如何规划？ ![微信图片_20240129090649](https://private-user-images.githubusercontent.com/149569575/300308295-5f032dd8-1eb2-4873-b28c-88c10690aff8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY5NzE1NDEsIm5iZiI6MTcxNjk3MTI0MSwicGF0aCI6Ii8xNDk1Njk1NzUvMzAwMzA4Mjk1LTVmMDMyZGQ4LTFlYjItNDg3My1iMjhjLTg4YzEwNjkwYWZmOC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUyOVQwODI3MjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNGViNWZmYWM3ODg0ZmU4OTM5NjAwNTIzZGVjNjkzMTg2YWVmMmRiMjEwOTdkOTllMjBkMzZkM2I0YTMwODc0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QMfdV0X7T6bgDEpgJPugZRi1dfYzyzHJ5ZdUMFCRefY)

The self.__modify_main_config() method in the tb_gateway_service.py file overrides the current tb_gateway configuration with environment variables