thingsboard/thingsboard-gateway #1233: [BUG] BACNet connector with multiple devices at same IP

elgutierrez, 2023-11-19:
**Describe the bug**
I'm using BACnet and I have one physical device in one IP/PORT and I'd like to separate it into multiple Thingsboard devices.

```
{
  "general": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1024,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "SCIRP_F3_CWP_EM1",
      "deviceType": "default",
      "address": "192.168.1.149:47808",
      "pollPeriod": 60000,
      "timeseries": [
        {
          "key": "kW",
          "objectId": "analogInput:1",
          "propertyId": "presentValue"
        },
        {
          "key": "kWh",
          "objectId": "analogInput:3",
          "propertyId": "presentValue"
        }
      ]
    },
    {
      "deviceName": "SCIRP_F3_CWP_BTUM1",
      "deviceType": "default",
      "address": "192.168.1.149:47808",
      "pollPeriod": 60000,
      "timeseries": [
        {
          "key": "kWporTR",
          "objectId": "analogInput:0",
          "propertyId": "presentValue"
        },
        {
          "key": "TR",
          "objectId": "analogInput:2",
          "propertyId": "presentValue"
        }
      ]
    },
    {
      "deviceName": "SCIRP_F3_CWP_FM1",
      "deviceType": "default",
      "address": "192.168.1.149:47808",
      "pollPeriod": 60000,
      "timeseries": [
        {
          "key": "vazao",
          "objectId": "analogInput:4",
          "propertyId": "presentValue"
        }
      ]
    },
    {
      "deviceName": "SCIRP_F3_CWP_TS1",
      "deviceType": "default",
      "address": "192.168.1.149:47808",
      "pollPeriod": 60000,
      "timeseries": [
        {
          "key": "tAAG",
          "objectId": "analogInput:5",
          "propertyId": "presentValue"
        },
        {
          "key": "tRAG",
          "objectId": "analogInput:6",
          "propertyId": "presentValue"
        }
      ]
    }
  ],
  "logLevel": "INFO",
  "name": "Bacnet Connector"
}
```

If I add just one of those devices, it works fine. As soon as I add more than one, I get an error

**Connector name (If bug in the some connector):**
BACNet

**Error traceback (If available):**
```
2023-11-19 08:18:47 - |ERROR| - [tb_logger.py] - tb_logger - exception - 97 - 'name'
Traceback (most recent call last):
  File "/home/enerlab-1/.local/lib/python3.9/site-packages/thingsboard_gateway/connectors/bacnet/bacnet_connector.py", line 283, in add_device
    device_name = device["deviceName"].replace("${" + device_name_tag + "}", data.pop("name"))
KeyError: 'name'
```

**Versions (please complete the following information):**
 - OS: Debian RPI4
 - Thingsboard IoT Gateway version 3.4.3
 - Python version 3.7

HarvUK, 2023-11-19:
This would be really useful - this was the next thing I want to do once I get BACnet working again.

PS. Are you sure you are on 3.4.3 - the latest release I can see is 3.4.2.

Harv

elgutierrez, 2023-11-20:
I thought that this was implemented. This kind of configuration works with Modbus so I assumed the same for BACnet. 

I'm actually using the `master` branch on the repo, as 3.4.2 is still have issues (#1213, #1219,#1224)  with bacnet and we need a release.

HarvUK, 2023-11-20:
I agree, no movement on #1225 as yet - does the latest branch work for you?

elgutierrez, 2023-11-20:
Yes, it works 👍

HarvUK, 2023-11-20:
Sorry to hijack your thread, but following the install guide  [here](https://thingsboard.io/docs/iot-gateway/install/source-installation/) I get this error when I run the last command of

`python3 ./thingsboard_gateway/tb_gateway.py`

```
s4444@s4444-Virtual-Machine:~/thingsboard-gateway$ python3 ./thingsboard_gateway/tb_gateway.py
Traceback (most recent call last):
  File "/home/s4444/thingsboard-gateway/./thingsboard_gateway/tb_gateway.py", line 17, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
ModuleNotFoundError: No module named 'thingsboard_gateway'
```
Am I missing a step? I've followed the instructions on both a PI and Ubuntu.

Thanks,

Harv

elgutierrez, 2023-11-21:
Yes, you should run `pip3 install .` in that same folder. I'm not a Python expert but it worked for me

elgutierrez, 2023-11-21:
And by the way, I think I fixed that issue just by changing this line:
`device_name = device["deviceName"].replace("${" + device_name_tag + "}", data.pop("name"))` to `device_name = device["deviceName"].replace("${" + device_name_tag + "}", data['name'])`. I just don't know what are the side effects that it could have, since I imagine that they do it so we don't match two times the same device

imbeacon, 2023-11-28:
Hi,

@elgutierrez Thank you for your contribution, it is a possible way to achieve expected behaviour and it should work properly, we have added the same to the master branch.

elgutierrez, 2023-11-28:
That's awesome, many thanks @imbeacon!

closed 2023-11-28 by elgutierrez

elgutierrez, 2023-11-28:
@imbeacon are the changes in the latest release?

imbeacon, 2023-11-28:
@elgutierrez at the moment they are in the master branch and will be available in the next release, we are going to fix some bugs and release.