thingsboard/thingsboard-gateway #1241: [HELP] thingsboard gateway modbus-rtu

KarimSayedAmir, 2023-11-26:
Hello together, 

I updated the Gateway to 3.4.3

I tried my best to solve the issue but i am stuck. 

I gett the following console ourput:
owipex_adm@owipexTest://var/log/thingsboard-gateway$ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-11-26 01:47:59 CET; 12s ago
   Main PID: 1452 (python3)
      Tasks: 34 (limit: 4416)
     Memory: 44.0M
        CPU: 6.366s
     CGroup: /system.slice/thingsboard-gateway.service
             └─1452 /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Nov 26 01:48:06 owipexTest python3[1452]: 2023-11-26 01:48:06 - |ERROR| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 515 - 'bool' object is not subscriptable
Nov 26 01:48:06 owipexTest python3[1452]: Traceback (most recent call last):
Nov 26 01:48:06 owipexTest python3[1452]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_gateway_remote_configurator.py", line 445, in _handle_connector_configuration_update
Nov 26 01:48:06 owipexTest python3[1452]:     config_file_name = config['configuration']
Nov 26 01:48:06 owipexTest python3[1452]: TypeError: 'bool' object is not subscriptable
Nov 26 01:48:06 owipexTest python3[1452]: 2023-11-26 01:48:06 - |ERROR| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 515 - 'bool' object is not subscriptable
Nov 26 01:48:06 owipexTest python3[1452]: Traceback (most recent call last):
Nov 26 01:48:06 owipexTest python3[1452]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_gateway_remote_configurator.py", line 445, in _handle_connector_configuration_update
Nov 26 01:48:06 owipexTest python3[1452]:     config_file_name = config['configuration']
Nov 26 01:48:06 owipexTest python3[1452]: TypeError: 'bool' object is not subscriptable

Any ideas?
Thanks for the effort.

**Versions (please complete the following information):**
 - OS: Ubuntu Server 22.03]
 - Thingsboard IoT Gateway version 3.4.3]
 - Python version3.10.12

imbeacon, 2023-11-27:
Hi @KarimSayedAmir,

Thank you for your interest in ThingsBoard IoT gateway.
Please try to make a backup of your configuration, remove the old gateway and install a version 3.4.3.1 it contains minor fixes for logging system and may help with this issue. Let us know about results.

KarimSayedAmir, 2023-11-27:
Hi @imbeacon, 

thanks a lot for the quick reply. I will check this today and give you feedback. 
Have a nice day.

KarimSayedAmir, 2023-11-27:
Hi @imbeacon, 

actually I am getting this now: 
owipex_adm@owipex:/boot/firmware$ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-11-27 21:54:54 CET; 2min 1s ago
   Main PID: 1764 (python3)
      Tasks: 29 (limit: 4416)
     Memory: 43.5M
        CPU: 45.666s
     CGroup: /system.slice/thingsboard-gateway.service
             └─1764 /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Nov 27 21:55:34 owipex python3[1764]:   File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 575, in read
Nov 27 21:55:34 owipex python3[1764]:     buf = os.read(self.fd, size - len(read))
Nov 27 21:55:34 owipex python3[1764]: TypeError: 'NoneType' object cannot be interpreted as an integer
Nov 27 21:55:34 owipex python3[1764]: 21:55:34 DEBUG modbus_connector:323 Device connector type: serial
Nov 27 21:55:34 owipex python3[1764]: 21:55:34 DEBUG modbus_connector:419 Modbus trying connect to Modbus Temperature Sensor
Nov 27 21:56:39 owipex python3[1764]: 21:56:39 INFO  modbus_connector:312 RS485_serial has been stopped.
Nov 27 21:56:44 owipex python3[1764]: 21:56:44 DEBUG modbus_connector:536 With result Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)
Nov 27 21:56:44 owipex python3[1764]: 21:56:44 DEBUG modbus_connector:359 Checking timeseries for device Modbus Temperature Sensor
Nov 27 21:56:44 owipex python3[1764]: 21:56:44 DEBUG modbus_connector:360 Device response:
Nov 27 21:56:44 owipex python3[1764]: 21:56:44 ERROR modbus_connector:340 Socket is closed
owipex_adm@owipex:/boot/firmware$

1.)Can this be connected to the Ubuntu 22.04 right management?

2.)Strange thing is that if I am using my own Modbus Lib, i can read the sensor easily. 

Have great evening
BR
Sayed

KarimSayedAmir, 2023-11-27:
Hi could solve the first issue: 

The installation was done over the deb package.
I created the missing folder:
`sudo` mkdir -p /var/lib/thingsboard_gateway/logs`

Then i had to add permissions: 

`sudo chown -R thingsboard_gateway:thingsboard_gateway /var/lib/thingsboard_gateway/logs
sudo chmod -R 755 /var/lib/thingsboard_gateway/logs`

Now the log looks like this: owipex_adm@owipex:/boot/firmware$ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2023-11-27 22:16:29 CET; 4min 37s ago
   Main PID: 1906 (python3)
      Tasks: 17 (limit: 4416)
     Memory: 42.8M
        CPU: 57.537s
     CGroup: /system.slice/thingsboard-gateway.service
             └─1906 /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

Nov 27 22:16:35 owipex python3[1906]: 2023-11-27 22:16:35 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_general_configuration_update - 327 - --- Checking other configuration >
Nov 27 22:16:35 owipex python3[1906]: 2023-11-27 22:16:35 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_general_configuration_update - 330 - --- Saving new general configurat>
Nov 27 22:16:35 owipex python3[1906]: 2023-11-27 22:16:35 - |ERROR| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_connector_configuration_update - 519 - 'bool' object is not subscript>
Nov 27 22:16:35 owipex python3[1906]: Traceback (most recent call last):
Nov 27 22:16:35 owipex python3[1906]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_gateway_remote_configurator.py", line 449, in _handle_connector_configuration_update
Nov 27 22:16:35 owipex python3[1906]:     config_file_name = config['configuration']
Nov 27 22:16:35 owipex python3[1906]: TypeError: 'bool' object is not subscriptable
Nov 27 22:16:35 owipex python3[1906]: 2023-11-27 22:16:35 - |INFO| - [modbus_connector.py] - modbus_connector - close - 312 - RS485_serial has been stopped.
Nov 27 22:16:35 owipex python3[1906]: 2023-11-27 22:16:35 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_grpc_configuration_update - 380 - Processed GRPC configuration update >
Nov 27 22:16:35 owipex python3[1906]: 2023-11-27 22:16:35 - |INFO| - [tb_gateway_remote_configurator.py] - tb_gateway_remote_configurator - _handle_storage_configuration_update - 355 - Processed storage configuration u>

But the sensor is still not reading. 
BR
Sayed

KarimSayedAmir, 2023-11-28:
Issue is now solved. 
Thanks a lot.

Tasosgi, 2024-01-03:
How? I have similar problem

closed 2024-06-20 by imbeacon