thingsboard/thingsboard-gateway #1247: [HELP] Missing documentation -  How to use RPC in Modbus connector

mrangics, 2023-12-02:
**Describe the issue**
I would like to use RPC to update registers on ModBus devices connected to ThingsBoard with Thingsboard Gateway.
There is a documentation available, but it's incomplete, please complete it.
https://thingsboard.io/docs/iot-gateway/guides/how-to-use-rpc-modbus-connector/

**Configuration** (Attach your configuration file)
[tb_gateway.json](https://github.com/thingsboard/thingsboard-gateway/files/13536677/tb_gateway.json)

**Connector name (If you need help with some connector/converter):**
[modbus_.json](https://github.com/thingsboard/thingsboard-gateway/files/13536678/modbus_.json)

**Error traceback (If it was raised):**
Request Time out.
```

**Versions (please complete the following information):**
 - OS: [Ubuntu 22.04]
 - Thingsboard IoT Gateway version [3.4.2]
 - Python version[3.11]

HarvUK, 2023-12-02:
Have a look at this [link](https://thingsboard.io/docs/iot-gateway/config/modbus/) think it might help?

Harv

mrangics, 2023-12-02:
@HarvUK , yes, my ModBus connector matched this, that's why I'm asking for help using the RPC, it looks like the definition is good in my connector - if you think not good, please orient me, I have uploaded my connector file.

Thanks,
Máté

mrangics, 2024-01-27:
Partially successful. RPC sometimes is being sent, sometimes not, more investigation is necessary.

pro100805, 2024-01-30:
Good afternoon, could you give a concrete modbus_.json for one device using rpc, the variant you have works