thingsboard/thingsboard-gateway #1254: [HELP] Remote Configuration: Too large message

devaskim, 2023-12-10:
**Describe the issue**
After enabling `remoteConfiguration` option I see a lot of the following messages in the gateway logs:
```
thingsboard-gateway  | 2023-12-10 08:15:53 - |DEBUG| - [tb_device_mqtt.py] - tb_device_mqtt - _on_disconnect - 205 - Disconnected client: <paho.mqtt.client.Client object at 0x7f8255c045d0>, user data: None, result code: 7
```
while in a server log
```
2023-12-10 08:16:23,228 [nioEventLoopGroup-4-12] ERROR o.t.s.t.mqtt.MqttTransportHandler - [e58f2d14-914c-47c7-92d8-180e40d38fd1] Message decoding failed: too large message: 148657 bytes
```
Then I set `NETTY_MAX_PAYLOAD_SIZE` server option to `1000000` and restart, but no success result
[image]

**Connector name (If you need help with some connector/converter):**
There is one connector of MQTT type with the following config. Gateway payload size is: `"maxPayloadSizeBytes": 1024000,`
[mqtt.json](https://github.com/thingsboard/thingsboard-gateway/files/13627210/mqtt.json)

**Versions (please complete the following information):**
 - TB server: 3.6.0
 - Gateway in Docker: 3.4.3.1

devaskim, 2024-01-30:
@imbeacon Any ideas?

imbeacon, 2024-01-30:
Hi @devaskim,

Disconnect event - it is OK in case of updating general configuration on the dashboard, but `2023-12-10 08:16:23,228 [nioEventLoopGroup-4-12] ERROR o.t.s.t.mqtt.MqttTransportHandler - [e58f2d14-914c-47c7-92d8-180e40d38fd1] Message decoding failed: too large message: 148657 bytes` - is an issue and we will investigate this point.

devaskim, 2024-02-21:
@imbeacon Any updates there ?

ashdam, 2024-08-05:
@imbeacon same problem here with the message size while trying to update from TB Edge PE adding a new connector with lots of timeseries.

ERROR o.t.s.t.mqtt.MqttTransportHandler - [6b2c0903-1ed6-4cad-9dde-888c5bc7369f] Message decoding failed: too large message: 485452 bytes

closed 2025-07-08 by imbeacon