thingsboard/thingsboard-gateway #1255: [BUG] get_env_variables() erroneously overrides file settings if running from PyCharm on Windows

devaskim, 2023-12-10:
**Describe the bug**
On Windows there is system environment variable called `USERNAME` (to test just run in command line `echo %USERNAME%), so after gateway start, [it scans ](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/gateway/tb_gateway_service.py#L111) environment variables and override gateway configuration specified in `tb_gateway.json`

imbeacon, 2024-01-30:
Hi @devaskim,

Thank you for information, please try the version from the master branch, new env variables were added for this case (Previous still present, for backward compatibility, but will be removed in the future releases):

    TB_GW_HOST
    TB_GW_PORT
    TB_GW_ACCESS_TOKEN
    TB_GW_CA_CERT
    TB_GW_PRIVATE_KEY
    TB_GW_CERT
    TB_GW_CLIENT_ID
    TB_GW_USERNAME
    TB_GW_PASSWORD

devaskim, 2024-02-21:
This doesn't fix the problem. Gateway [still reads](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/gateway/tb_gateway_service.py#L113) from env variable called `username`. It leads to overriding the valid credentials in `tb_gateway.json`

imbeacon, 2024-02-21:
Old variables will be removed in one of the future releases.

closed 2024-02-21 by devaskim