thingsboard/thingsboard-gateway #1258: BACnet device (Room Control Simulator) not connected to ThingsBoard Platform

kamisettychandan, 2023-12-11:
Hi Guys
I am new to ThingsBoard Platform, can anyone guide me

Aim: I want to send Room Control Simulator BACnet device to ThingsBoard Platform.

**Successful steps:**
--------------------
      1. Provision Gateway in Things Board Platform - Successful
      2. Install Things Board Gateway in Local Machine using Docker - Successful
      3. Establish Connection between Things Board Gateway & Things Board Platform (using Access Token)- Successful
      4. Add BACnet Connectors to the Main Configuration file - Successful
      5. Configure BACnet Connector - Successful
      6. Provision Device in Things Board Platform - Successful
      7. Connect Device to Things Board Gateway (mapped to specific Gateway) - Successful
      8. Check Device Connectivity for sending telemetry on behalf of the device using shell (HTTP & MQTT) - Successful
      9. Configure Dashboard for Visualization (behalf of the device using shell (HTTP & MQTT)) - Successful
      10.Configure Alarms (behalf of the device using shell (HTTP & MQTT)) - Successful  
      11.Observe Room Control Simulator device data in YABE - Successful
          
**Unsuccessful steps:**
------------------------
1. Simulator data not reached to ThingsBoard Platform

bacnet.json:
------------
{
  "general": {
    "objectName": "TB_gateway",
    "address": "127.0.0.1:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.55.103:47808",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "type": "string",
          "objectId": "analogOutput:1",
          "propertyId": "presentValue"
        }
        
Kindly guide me...!

devaskim, 2023-12-18:
Try to use `192.168.55.103` instead of `127.0.0.1`

kamisettychandan, 2023-12-18:
Hi,
I used IP Address 192.168.55.103 but Device not connected to ThingsBoard Platform.
I added device ("provisionDeviceKey", "provisionDeviceSecret) in tb_gateway.json file.

**bacnet.json:**
{
  "general": {
    "objectName": "TB_gateway",
    "address": "192.168.55.103:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.55.103:52509",
      "pollPeriod": 10000,
      "attributes": [
        {
          "key": "temperature",
          "type": "string",
          "objectId": "analogOutput:1",
          "propertyId": "presentValue"
        }
      ],
      
      
      
   **tb_gateway.json:**
    {
  "thingsboard": {
    "host": "thingsboard.cloud",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "oxoIy56YHKB8DhsARqr8"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
  {
	  "name":"BACnet Connector",
	  "type":"bacnet",
	  "configuration":"bacnet.json"
  }
  ],
  "provisioning":{
	  "type":"AUTO",
	  "provisionDeviceKey":"2kbm4q67m7hajkj8laop",
	  "provisionDeviceSecret":"551hvcxubptiz92zfbkk"
  }
}

thiscantbeserious, 2024-01-17:
actually this might becrelated to how bacpypes works or more deeply the network stack below it.

TLDR: 

Use 0.0.0.0:47808 for your gateway address

Using 0.0.0.0 will enable it to transmit on any interface no matter the netmask

if you're using docker you might also want to use host network mode and possibly run the container in privileged mode if that still doesn't help

kamisettychandan, 2024-01-18:
I have did like that but not connected
{
  "general": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.55.103:52509",
      "pollPeriod": 10000,
      
      Kindly guide me

sergeybuza, 2024-01-29:
The same thing. My config:
1. Thingsboard Gatewy on Laptop_1
2. Thingsboard platform in Cloud
3. RoomSimulator and Yabe via mono in Laptop_2

Current situation:
See greating exchange between TB-GW and RoomSimulator and nothing else more. Any ideas?

[image]

@kamisettychandan 
Had you forward port in Docker?

kamisettychandan, 2024-02-12:
Hi,

Sorry for late replay.

1. ThingsBoard Gateway configured in my personal laptop & RoomSimulator & Yabe also working in same personal laptop only.
2. I didn't forward port in Docker.

Please, help me out

kamisettychandan, 2024-02-23:
Hi @samson0v @imbeacon,

I am not able to send Room Control Simulator data to Things Board Platform. 
Please help me out.

kamisettychandan, 2024-03-07:
@sergeybuza 

Kindly help me...!
As per Docker Logs, BACnet connector not detecting 'Room Control Simulator' device. Could you help me...

Please find logs:
[Day 4 Docker_Logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523215/Day.4.Docker_Logs.txt)

Please find config logs:
[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523193/bacnet.json.txt)
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523194/tb-gateway.txt)

kamisettychandan, 2024-05-09:
@sergeybuza,

Kindly help me out

imbeacon, 2025-04-16:
The latest version and few previous versions of the gateway can easily connect to Room control simulator.