thingsboard/thingsboard-gateway #1266: [HELP WANTED] Data Already successfully send in TB device (Gateway), but my sensor still cannot get telemetry data

yinghai97, 2023-12-20:
I able to run thingsboard-gateway in CLI
[image]

and I successfully get some info from gateway 
[image]

but I the data did not send to my sensor device
[image]

Am I do something incorrectly? 

[modbus.json](https://github.com/thingsboard/thingsboard-gateway/files/13726451/modbus.json)
[tb_gateway.json](https://github.com/thingsboard/thingsboard-gateway/files/13726452/tb_gateway.json)

yinghai97, 2023-12-20:
[image]
I solved the problem by add this in shared attribute.
But I not sure whether is correct way.

you may close this post if this is the correct way
thankyou

DuXiaoChuang, 2024-01-30:
[image]
[image]
 My question is the same as yours，"Can you help me take a look at my question, just like yours?"

imbeacon, 2024-06-19:
Closed, due to inactivity, should be solved in one of the latest releases.

closed 2024-06-19 by imbeacon