thingsboard/thingsboard-gateway #1277: [BUG] Latest version (3.4.4) installation on RPI is failing: missing python3-cryptography

elgutierrez, 2024-01-11:
**Describe the bug**
I'm trying to install the latest gateway version on a RPI4 using both Raspbian and Ubuntu using the `deb` package. On both cases I'm getting a missing dependency (`python3-cryptography 3.5.0`) and I don't know how to fix it. I ran `apt update` and `apt upgrade` but no success. 

**Error traceback (If available):**
```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'python3-thingsboard-gateway' instead of './python3-thingsboard-gateway.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python3-thingsboard-gateway : Depends: python3-cryptography (< 3.5) but 38.0.4-3 is to be installed
E: Unable to correct problems, you have held broken packages.
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20 and Debian 12
 - Thingsboard IoT Gateway version 3.4.4
 - Python version 3.11.2

Allisam-c, 2024-01-12:
I also ran into this problem, but in the end I didn't solve it and opted to use docker for the installation instead.

elgutierrez, 2024-01-12:
I was using the Docker container but[ I faced issues on v3.4.2](https://github.com/thingsboard/thingsboard-gateway/issues/1223). I then switched to an installation from the sources because [I had to patch a bug on the Bacnet connector](https://github.com/thingsboard/thingsboard-gateway/issues/1233#issuecomment-1830276042). And now this. It's a little bit concerning the fact that we don't have stable releases, and on each realease we fix many bugs but then we introduce many others.

Other than that, I'm seeing issues with the remote logging option and issues with remote configuration. I'll open new issues for them

Tasosgi, 2024-01-22:
Try the Ubuntu Server 22.04.3 LTS. I had the same issue.

enyulion, 2024-01-30:
Openssl update to 3.x
All older versions (including 1.1.1, 1.1.0, 1.0.2, 1.0.0 and 0.9.8) are now out of support and should not be used.

AxelGard, 2024-06-17:
Im having the same issue is the only solution to build cryptography? Or have someone found a solution?

AxelGard, 2024-06-17:
> Im having the same issue is the only solution to build cryptography? Or have someone found a solution? 

Ohh 😮 this might fix the issue I hope https://github.com/thingsboard/thingsboard-gateway/blame/b108d9b6795f4d745f60feb1db27c57c7b5f7d16/setup.py#L56C5-L56C10

imbeacon, 2024-06-20:
Hi,

Please try to use the latest release deb file. We have removed fixed cryptography version dependency from the gateway, but it may produce some errors with OPC-UA connector in case of using too old version.

If you encounter this issue again, you may try to install rust compiler and it will build cryptography correctly, to do this please try to use the following command (Should work for RPi, for ubuntu you need to change default-host to x86_64-unknown-linux-gnu):

curl https://sh.rustup.rs -sSf | sh -s -- -y --default-host=armv7-unknown-linux-gnueabihf

closed 2024-07-01 by imbeacon