thingsboard/thingsboard-gateway #1279: [FEATURE REQUEST] Modbus-tcp connection limit issue

Allisam-c, 2024-01-12:
**Describe the issue**
I am currently using the thingsboard-gateway to collect telemetry data from my equipment, using RTUs from the oil field. the problem I am having is that each rtu only allows a maximum of 16 simultaneous connections to exist, and some rtu's even only support 10 connections, and one rtu is using multiple unitId's on a single ip, port. this is leading to my can only connect to some of the devices, and the connection is lost for devices that exceed the maximum number of connections.

The result of the test now is that each slave with the same ip, port, and unitId in modbus.json takes up one connection, and the current rtu is configured with multiple unitId's, exceeding the maximum number of connections.

**Versions (please complete the following information):**
 - OS: [e.g. Centos 7.9]
 - Thingsboard IoT Gateway version [e.g. 3.4.3.1]
 - Python version[e.g. 3.9]

[image]

JimmyPedersen, 2024-06-03:
I have the same problem with some of my modbus slaves.
One type can only have a max of 3 simultaneous connections and currently spits out a lot of Connection lost messages.
Please add this feature!

imbeacon, 2025-04-16:
This issue was solved in previous releases, please use the latest release to get this and other features and bug fixes.

closed 2025-04-16 by imbeacon