thingsboard/thingsboard-gateway #1286: [HELP] Gateway disconnect and connect every some msec

Tasosgi, 2024-01-28:
I can sent telemetry data to platform but the gateway disconnect and connect every some msec. It seems that it tries to subscribe to attributes without success.

[image]

[modbus_serial.txt](https://github.com/thingsboard/thingsboard-gateway/files/14076826/modbus_serial.txt)
[tb_gateway.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14076827/tb_gateway.json.txt)

imbeacon, 2024-01-29:
Hi @Tasosgi,

Usually disconnect with this reason code appears if you have already connected gateway with the same credentials, could you try to generate a new accessToken and use it?

Tasosgi, 2024-01-29:
Hello, yes I have seen that in other cases. But in this case there isn't another gateway or device online. Anyway I will check it today.

Another strange think I see in this version is that if I don't put the "type": "serial" **before** "master" my configuration file don't works and the gateway use the default one. According to the examples it must  be under "slaves". I have attach my configuration files

imbeacon, 2024-01-29:
Ok, another option is rate limits, could you check your notifications on the platform - don't you have a message like "Rate limits exceeded?

Tasosgi, 2024-01-29:
No message like this. This platform hasn't big load. There was a traffic yesterday when I tested the gateway but I think because the repeatedly connect - disconnect procedure.

 
[image]

imbeacon, 2024-01-30:
@Tasosgi I mean notification (bell icon in the top panel), there is rate limits per device on ThingsBoard level. And if you restart the gateway 3 or 4 times per minute - it may be triggered.

Tasosgi, 2024-02-06:
Thanks to the help of @imbeacon I found the problem. I had gave the same name to the gateway, when I created it in the platform, and the slave, in the modbus configuration file

closed 2024-02-06 by Tasosgi