thingsboard/thingsboard-gateway #1288: [HELP] connected to external MQTT Broker

langlichong, 2024-01-30:
**Describe the issue**
does gateway support to connect to a third-party MQTT Broker ,such as mosquitto  , so to integrate with other system other than thingsboard itself

devaskim, 2024-01-30:
@langlichong Did you read [the docs](https://thingsboard.io/docs/iot-gateway/config/mqtt/) of MQTT connector of the gateway?

> This guide will help you to get familiar with MQTT Connector configuration for ThingsBoard IoT Gateway. Use [general configuration](https://engix-ltd.github.io/docs/iot-gateway/configuration/) to enable this Connector. The purpose of this Connector is to connect to external MQTT broker and subscribe to data feed from devices. Connector is also able to push data to MQTT brokers based on the updates/commands from ThingsBoard.

imbeacon, 2024-01-30:
Hi @langlichong,

To connect and collect data from the external broker you can use MQTT connector, as @devaskim mentioned. But if you want to send a collected data to some platform, except ThingsBoard, it is not possible to do with a gateway, because it uses special API to communicate with a platform. If you need to send somewhere received data from ThingsBoard - you can configure your rule chain and use MQTT rule node to send collected data to another broker.

langlichong, 2024-01-30:
thanks @devaskim  &  @imbeacon  !

closed 2024-01-30 by imbeacon