thingsboard/thingsboard-gateway #1290: [HELP]

star7128, 2024-01-30:
**Describe the issue**
If field devices only support TCP server communication, can we run TCP client based on existing gateway design?

Somehow like reverse socket connector if my understanding to tb-gateway is correct ...

imbeacon, 2024-02-09:
Hi @star7128,

Thank you for your interest in Thingsboard IoT Gateway, you are right, it may be like a reverse tcp connector, or additional configuration parameter for tcp connector like "role", we will add this feature request to a roadmap.

star7128, 2024-02-19:
Hi @imbeacon 

It is definitely helpful to be more compatible with field devices. Where to review thingsboard-gateway roadmap? We do not see the roadmap from the repository. Thanks!