thingsboard/thingsboard-gateway #1293: [HELP] How to add ocpp charger using thingsboard gateway

Vaghulkumar95, 2024-02-05:
**Describe the issue**
I'm trying to connect an EV Charger using ocpp and I'm unable to connect, please help.
I mainly need to know how to configure the host and port in the charger. And how to set the endpoint? 
[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
`thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: true
  remoteConfiguration: false
  maxPayloadSizeBytes: 1024
  minPackSendDelayMS: 0
  checkConnectorsConfigurationInSeconds: 60
  handleDeviceRenaming: true
  checkingDeviceActivity:
    checkDeviceInactivity: false
    inactivityTimeoutSeconds: 120
    inactivityCheckPeriodSeconds: 10
  security:
    accessToken: 
storage:
#  type: memory
#  read_records_count: 100
#  max_records_count: 100000
  type: file
  data_folder_path: ./data/
  max_file_count: 10
  max_read_records_count: 10
  max_records_per_file: 10000
connectors:

#   -
#    name: MQTT Broker Connector
#    type: mqtt
#    configuration: mqtt.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
   -
     name: OCPP Connector
     type: ocpp
     configuration: ocpp.json
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector`

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
ocpp.json
`{
  "centralSystem": {
    "name": "Central System",
    "host": "192.168.178.92",
    "port": 502,
    "connection": {
      "type": "insecure"
    },
      "security": []
  },
  "chargePoints": [
    {
      "idRegexpPattern": "charge_points/CP_1",
      "deviceNameExpression": "IoCharger PDF-22S",
      "deviceTypeExpression": "default",
      "attributes": [
        {
          "messageTypeFilter": "MeterValues,",
          "key": "temp1",
          "value": "Energy.Active.Import.Register,Power.Active.Import"
        },
        {
          "messageTypeFilter": "MeterValues,",
          "key": "vendorId",
          "value": "A10221013421"
        }
      ],
      "timeseries": [
        {
          "messageTypeFilter": "DataTransfer,",
          "key": "temp",
          "value": "${data.temp}"
        }
      ],
      "attributeUpdates": [
        {
          "attributeOnThingsBoard": "shared",
          "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
        }
      ],
      "serverSideRpc": [
        {
          "methodRPC": "rpc1",
          "withResponse": true,
          "valueExpression": "${params}"
        }
      ]
    }
  ]
}`

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```
`Feb 05 15:59:58 rock-3c python3.10[2720]:     self.__loop.run_until_complete(self.start_server())
Feb 05 15:59:58 rock-3c python3.10[2720]:   File "/usr/local/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
Feb 05 15:59:58 rock-3c python3.10[2720]:     return future.result()
Feb 05 15:59:58 rock-3c python3.10[2720]:   File "/usr/local/lib/python3.10/site-packages/thingsboard_gateway/connectors/ocpp/ocpp_connector.py", line 109, in start_server
Feb 05 15:59:58 rock-3c python3.10[2720]:     self._server = await websockets.serve(self.on_connect, host, port, subprotocols=['ocpp1.6'],
Feb 05 15:59:58 rock-3c python3.10[2720]:   File "/root/.local/lib/python3.10/site-packages/websockets/legacy/server.py", line 1116, in __await_impl__
Feb 05 15:59:58 rock-3c python3.10[2720]:     server = await self._create_server()
Feb 05 15:59:58 rock-3c python3.10[2720]:   File "/usr/local/lib/python3.10/asyncio/base_events.py", line 1519, in create_server
Feb 05 15:59:58 rock-3c python3.10[2720]:     raise OSError(err.errno, 'error while attempting '
Feb 05 15:59:58 rock-3c python3.10[2720]: OSError: [Errno 99] error while attempting to bind on address ('192.168.178.92', 502): cannot assign requested address`

**Versions (please complete the following information):**
 - OS: [e.g. Debian 11]
 - Thingsboard IoT Gateway version [3.3]
 - Python version[e.g. 3.10]

imbeacon, 2024-02-06:
Hi @Vaghulkumar95,

Please try to use something like 0.0.0.0 as a host and port more that 502 (e.g. 5020), for your connector, because in linux you should have root access to bind ports in range 1-1024.

Vaghulkumar95, 2024-02-07:
Hi @imbeacon,

[image]

[image]

Can you please confirm if I have set the properties correctly, still I'm not getting anything

Vaghulkumar95, 2024-02-09:
Hi @imbeacon, 

Just wanted to follow up on this? We are trying to get this EV Charger connected and client is waiting for the connection results. Any inputs on this is highly appreciated.

Vaghulkumar95, 2024-02-15:
Hi @imbeacon,

I'm now able to communicate with the battery. I had to give the domain as the hub's IP address and port as 5020, and in json I did give 0.0.0.0 and port as 5020 and it was able to connect. Can you let me know how to get the status notification, and hearbeat in the json? What would be the message filter type for these?

Vaghulkumar95, 2024-08-29:
HI @imbeacon @samson0v ,

Any updates on this?

closed 2026-09-10 by samson0v