thingsboard/thingsboard-gateway #1294: [HELP] Can't discover BACnet devices - Version 3.4.4

TomAllen95, 2024-02-06:
**Describe the issue**
Create description about your issue, and your actions to solve it.

 Have set up the BACnet connector from my ThingsBoard cloud instance, but it's not discovering the devices on the network. The devices are both at 192.168.11.35:47808, with first device being the gateway that connects to the plug via wireless mesh, with both devices being discoverable on YABE from the desktop machine. The command I used to set up the container is below, minus the host and accessToken
 
 `docker run -d -it --network=host --expose 8082 -e host= -e port=1883 -e accessToken= -p 60000-61000:60000-61000 -v "C:\Users\Office Demos\Documents\tb-gateway\config:/thingsboard_gateway/config" -v "C:\Users\Office Demos\Documents\tb-gateway\extensions:/thingsboard_gateway/extensions" -v "C:\Users\Office Demos\Documents\tb-gateway\logs:/thingsboard_gateway/logs" --name tb-gateway --restart always thingsboard/tb-gateway`

[image]

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

{
  "thingsboard": {
    "host": "thingsboard.cloud",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600,
      "configuration": null,
      "commands": []
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "accessToken": ""
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
    {
      "type": "bacnet",
      "name": "BACnet",
      "configuration": "bacnet.json"
    }
  ]
}

**Connector name (If you need help with some connector/converter):**
BACnet Connector

{
  "name": "BACnet",
  "logLevel": "DEBUG",
  "general": {
    "objectName": "TB_gateway",
    "address": "0.0.0.0:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1476,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "S_1AC0_DemoPlug",
      "deviceType": "default",
      "address": "192.168.11.35:47808",
      "pollPeriod": 10000,
      "timeseries": [
        {
          "key": "InsRealPower1",
          "type": "int",
          "objectId": "analogInput:2",
          "propertyId": "InsRealPower1"
        }
      ]
    },
    {
      "deviceName": "HoneywellGateway_MainOffice",
      "deviceType": "default",
      "address": "192.168.11.35:47808",
      "pollPeriod": 10000
    }
  ]
}

**Error traceback (If it was raised):**

**Versions (please complete the following information):**
 - OS: Windows 10 - Docker Desktop v 24.0.7
 - Thingsboard IoT Gateway version  3.4.4
 - Python version - 3.11.7

kamisettychandan, 2024-03-07:
@TomAllen95,
Hi, I am also facing same issue

imbeacon, 2024-03-07:
Hi @TomAllen95 @kamisettychandan,

Thank you for interest in ThingsBoard IoT gateway. At the moment we are investigating the connection ability to YABE simulator, unfortunately for now, with no luck, but we will write you after finding the reason why it cannot connect.

kamisettychandan, 2024-03-08:
@imbeacon 

Please find my docker logs, then you can understand the problem.

[Day.4.Docker_Logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/14532888/Day.4.Docker_Logs.txt)

Please find Config files,
[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14532894/bacnet.json.txt)
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/14532895/tb-gateway.txt)

Kindly check and update

mgruenberger21, 2024-09-19:
Hello,

any news on this topic?
I have the same issue.
TB-Gateway is sending "Unconfirmed-REQ who-Is"-Messages to 'Room Control Simulator'.
But nothing more happens.
[image]
I opended another case a few weeks ago, because I can`t access data from other bacnet-devices as well.
(https://github.com/thingsboard/thingsboard-gateway/issues/1520)
I can access all the data witch Yabe (of room simulator and bacnet end-device).

Is anyone here using bacnet-connector of tb-gateway successfully?

Kind regards.

imbeacon, 2024-12-11:
Hi all,

Please try to use the latest version of the gateway(https://github.com/thingsboard/thingsboard-gateway/releases/tag/3.6.2), we have completely remastered BACnet connector, migrate to bacpypes3 as a base for it, instead of bacpypes. Also, we have tested the connector with YABE, it was able to read and write data. 
I close this issue, because it relates to previous implementation of the connector, but please feel free to open new issues in case of noticing some misbehaving.

closed 2024-12-11 by imbeacon