thingsboard/thingsboard-gateway #1295: [HELP] preserve JSON type in attributeUpdates for MQTT

Fishwaldo, 2024-02-08:
**Describe the issue**
Is there a way to preserve the json type of a attribute when sending via the MQTT broker?

eg, if I define two shared attributes on my device:
test1 - string
test2 - integer

and modify those values, then the attributeValue is either only enclosed in quotes or not, depending upon the valueExpression in the in the attributeUpdates block of my configuration:
```
   {
      "retain": false,
      "deviceNameFilter": "SN.*",
      "attributeFilter": "uploadFrequency",
      "topicExpression": "sensor/${deviceName}/${attributeKey}",
      "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
    }
```

what I would like to do, it only add the quotes around the attributeValue if its a string, and not for other json types (eg, integer or float)

imbeacon, 2024-02-09:
Hi @Fishwaldo,

Thank you for your interest in ThingsBoard IoT gateway, it is a bug, we will fix it.

Fishwaldo, 2024-02-09:
Thanks. I believe the same issue will exist with the attributeRequests as well? (as in the reply posted back to a request)

imbeacon, 2024-02-09:
Probably yes, because we are using strings to form payload, we will add additional conversion to MQTT connector to avoid this situation.

imbeacon, 2024-02-12:
Hi @Fishwaldo,

We have added a fix for this in the master branch, you may try it or wait for the release.

Fishwaldo, 2024-02-12:
Hi,
Sorry, I don't see the commit where it's fixed?

imbeacon, 2024-02-12:
You may check on_attribute_update method in Matt_connector in bdbf1807.

closed 2024-02-26 by imbeacon