thingsboard/thingsboard-gateway #1299: [BUG] Config incorrect for mqtt

devaskim, 2024-02-11:
**Describe the bug**

After pulling out the latest source code (11 Jan) I faced with a problem of invalid configuration of each my mqtt connector.

There are magic lines of code in `tb_gateway_service.py` which my configurations cannot pass. The debugger pointed me out to an absent `logLevel`. Where this field should be set?

```
if ("logLevel" in connector_config["config"][config] and "name" in connector_config["config"][config] and len(connector_config["config"][config].keys()) > 3) or \
    ("logLevel" not in connector_config["config"][config] and "name" not in connector_config["config"][config] and len(connector_config["config"][config].keys()) > 1)
```

Here is one of connector's configuration:
```
{
  "broker": {
    "name": "ADFWEB",
    "host": "127.0.0.1",
    "port": 1883,
    "clientId": "ADFWebGateway",
    "maxMessageNumberPerWorker": 10,
    "maxNumberOfWorkers": 100,
    "sendDataOnlyOnChange": false,
    "sendDataOnlyOnChangeTtl": 10800000,
    "security": {
      "type": "basic",
      "username": "user",
      "password": "qwerty"
    }
  },
  "mapping": [
    {
      "topicFilter": "ADF_Web/HD67933/T_Flow",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "HD67933",
        "deviceTypeTopicExpression": "ADFWEB",
        "attributes": [],
        "timeseries": [
          {
            "type": "double",
            "key": "T_Flow",
            "value": "${value}"
          }
        ]
      }
    }
  ],
  "connectRequests": [],
  "disconnectRequests": [],
  "attributeRequests": [],
  "attributeUpdates": [],
  "serverSideRpc": [],
  "id": "93cc2506-adcd-43cb-ae48-d454770de1f1"
}
```
And here is snippet from `tb_gateway.json`
```
,
  "connectors": [
    {
      "name": "ADFWEB",
      "type": "mqtt",
      "configuration": "adfweb.json"
     }
  ]
```

imbeacon, 2024-02-12:
Hi @devaskim,

Please try the version from the master branch, bda8f12 - should fix the issue.

devaskim, 2024-02-21:
fixed, thanks

closed 2024-02-21 by devaskim