thingsboard/thingsboard-gateway #1305: [HELP] Reliability after gateway reboot - connectors won't start

vicuzz, 2024-02-21:
**Describe the issue**
When i reboot the gateway or restart the service, all configured connectors won't start autonomously
I have to manually deactivate and reactivate each connector in the dashboard for sending telemetry again.

**Configuration** (Attach your configuration file)
{
  "thingsboard": {
    "host": " ",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": false,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 50,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": " "
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 10,
      "inactivityCheckPeriodSeconds": "30"
    },
    "ts": 1708507364014
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7,
    "ts": 1708507364014
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000,
    "ts": 1708507364014
  },
  "connectors": [
    {
      "type": "modbus",
      "name": "test rtu",
      "configuration": "testRtu.json"
    }
  ]

**Connector name (If you need help with some connector/converter):**
every configured connectors on the gateway, example:

{
  "master": {
    "slaves": [
      {
        "type": "serial",
        "method": "rtu",
        "port": "/dev/ttyUSB0",
        "baudrate": 9600,
        "timeout": 35,
        "stopbits": 1,
        "bytesize": 8,
        "byteOrder": "LITTLE",
        "wordOrder": "LITTLE",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 5,
        "deviceName": "TEST RTU",
        "attributesPollPeriod": 5000,
        "timeseriesPollPeriod": 5000,
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 50,
        "waitAfterFailedAttemptsMs": 3000,
        "attributes": [
          {
            "tag": "string_read",
            "type": "string",
            "functionCode": 3,
            "objectsCount": 4,
            "address": 50000
          }
        ],
        "timeseries": [
          {
            "tag": "Tensione",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 50520
          },
          {
            "tag": "Frequenza",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 50526
          },
          {
            "tag": "Corrente",
            "type": "32uint",
            "functionCode": 3,
            "registerCount": 2,
            "address": 50528
          }
        ],
        "attributeUpdates": [],
        "rpc": []
      }
    ]
  },
  "logLevel": "INFO",
  "name": "test rtu"

**Error traceback (If it was raised):**
Sometimes this error is raised in the console after service restart, i have attached a full log after a service restart
```
Feb 21 10:31:01 trakkio python3[1710]: 2024-02-21 10:31:01 - |ERROR| - [tb_logger.py] - tb_logger - exception - 97 - 'NoneType' object cannot be interpreted as an integer
Feb 21 10:31:01 trakkio python3[1710]: Traceback (most recent call last):
Feb 21 10:31:01 trakkio python3[1710]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 347, in __process_slaves
Feb 21 10:31:01 trakkio python3[1710]:     input_data = self.__function_to_device(device, current_data)
Feb 21 10:31:01 trakkio python3[1710]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 514, in __function_to_device
Feb 21 10:31:01 trakkio python3[1710]:     result = device.config['available_functions'][function_code](address=config[ADDRESS_PARAMETER],
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/client/mixin.py", line 118, in read_holding_registers
Feb 21 10:31:01 trakkio python3[1710]:     return self.execute(request)
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/client/base.py", line 184, in execute
Feb 21 10:31:01 trakkio python3[1710]:     return self.transaction.execute(request)
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/transaction.py", line 168, in execute
Feb 21 10:31:01 trakkio python3[1710]:     response, last_exception = self._transact(
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/transaction.py", line 323, in _transact
Feb 21 10:31:01 trakkio python3[1710]:     result = self._recv(response_length, full)
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/transaction.py", line 362, in _recv
Feb 21 10:31:01 trakkio python3[1710]:     read_min = self.client.framer.recvPacket(min_size)
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/framer/rtu_framer.py", line 323, in recvPacket
Feb 21 10:31:01 trakkio python3[1710]:     result = self.client.recv(size)
Feb 21 10:31:01 trakkio python3[1710]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.10/site-packages/pymodbus/client/serial.py", line 302, in recv
Feb 21 10:31:01 trakkio python3[1710]:     result = self.socket.read(size)
Feb 21 10:31:01 trakkio python3[1710]:   File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 575, in read
Feb 21 10:31:01 trakkio python3[1710]:     buf = os.read(self.fd, size - len(read))
Feb 21 10:31:01 trakkio python3[1710]: TypeError: 'NoneType' object cannot be interpreted as an integer

```

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS on raspberry Pi 4
 - Thingsboard IoT Gateway version 3.4.4
 - Python version 3.10.12
[restart-log.txt](https://github.com/thingsboard/thingsboard-gateway/files/14357451/restart-log.txt)

samson0v, 2024-02-28:
Hi @vicuzz,
Can you, please, provide more information about your issue:
1. What type of connectors do you configure and how many connectors do you try to start at the same time?
2. Does the exception that you mentioned above appear every time or only when the connector is restarted?

If you using several connectors, please, provide their configurations.
Also, send, please, the full log file.

vicuzz, 2024-03-12:
Yes, solved after upgrade to latest release
Thanks!

closed 2024-03-12 by vicuzz