thingsboard/thingsboard-gateway #1312: [HELP] BACnet devices not connected to TB Platform - Version 3.4.4 #1294[HELP]

kamisettychandan, 2024-02-26:
**Describe the issue**
Create description about your issue, and your actions to solve it.

I want to push Room Control Simulator (BACnet device) telemetry to TB Platform.
for this,

1. Installed TB-Gateway in local machine (windows 10) using Docker
2. Updated TB-Gateway Access Token in tb-gateway.json then Gateway Device came into ACTIVE state
3. Now, provisioned Device in TB Platform to get Telemtery data from Room Control Simulator
4. I am observing Simulator data in YABE Application tool
5. But, Telemetry not discovered in TB-Platform

  
**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
 Connector Name: BACnet Connector

**Versions (please complete the following information):**
 - OS: Windows 10
 - Thingsboard IoT Gateway version 3.4.4
 - Docker Version: 25.0.2

[tb-gateway.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14403298/tb-gateway.json.txt)
[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14403299/bacnet.json.txt)

Kindly help me out...!

kamisettychandan, 2024-03-07:
@ouertani @wayyen @imbeacon,

Kindly help me...!
As per Docker Logs, BACnet connector not detecting 'Room Control Simulator' device. Could you help me...

Please find logs:
[Day 4 Docker_Logs.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523215/Day.4.Docker_Logs.txt)

Please find config logs:
[bacnet.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523193/bacnet.json.txt)
[tb-gateway.txt](https://github.com/thingsboard/thingsboard-gateway/files/14523194/tb-gateway.txt)

imbeacon, 2025-01-15:
Hi @kamisettychandan,

The BACnet connector was completely refactored and doesn’t have issues now with YABE recognition. Please try to use the latest version of the gateway.

closed 2025-01-15 by imbeacon