thingsboard/thingsboard-gateway #1328: tb_device_mqtt on_connect  connection FAIL with error Bad user name or password Bad user name or password

struggle3, 2024-03-06:
**Describe the bug**
Cannot load GRPC connector!
2024-03-06 13:42:32 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 218 - Gateway starting...
2024-03-06 13:42:32 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 223 - ThingsBoard IoT gateway version: 0.0
2024-03-06 13:42:32 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 309 - Gateway started.
2024-03-06 13:42:32 - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import ModbusConnector from E:\ThingsBoard\src\thingsboard-gateway\thingsboard_gateway\connectors\modbus.
2024-03-06 13:42:34 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 372 - connection FAIL with error Bad user name or password Bad user name or password
2024-03-06 13:42:34 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 181 - Subscribed to *|* with id 1 for device *
2024-03-06 13:42:38 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 84 - 3.4.5
2024-03-06 13:42:38 - |INFO| - [tb_updater.py] - tb_updater - check_for_new_version - 86 - 

[===UPDATE===]

 New version 3.4.5 is available! 

[===UPDATE===]

**Versions (please complete the following information):**
 - OS: [e.g. Windows 11]
 - Thingsboard IoT Gateway version 3.4.5
 - Python version 3.8

Allisam-c, 2024-05-14:
thingsboard_gateway\gateway\tb_gateway_service.py
# change main config if Gateway running with docker env variables
self.__modify_main_config()   <---Annotate this line
---
"Environment variables have more priority than configuration parameters. If you don't pass the variable, the gateway will not use it and it will use values from the configuration file."---issues#1232
The meaning is that in the development environment, the system's environment variables are read here, not from the configuration file.

struggle3, 2024-05-14:
> thingsboard_gateway\gateway\tb_gateway_service.py
>
> # change main config if Gateway running with docker env variables
> ## self.__modify_main_config() <---Annotate this line
> "Environment variables have [...]

After that, I commented the code. When running cmd, I still cannot connect to the mqtt broker normally.

imbeacon, 2024-06-06:
Hi @struggle3,

This issue is fixed in the master branch, you can use it or wait for the release, it is expecting to be at the end of June.

closed 2024-06-19 by imbeacon