thingsboard/thingsboard-gateway #1334: [BUG] ARM64 docker image 3.4.5 won't start: exec format error

wz2b, 2024-03-12:
**Describe the bug**
Installing docker image thingsboard/tb-gateway:latest (which is 3.4.5) on an aarch64 platform correctly pulls the arm64 image, but it won't start.  The container logs report `exec /bin/sh: exec format error`.  Typically this means that it's not the correct binary for that architecture.  I looked at the dockerfile and I don't see an obvious problem so I am not exactly sure what's going on here.

**Versions (please complete the following information):**
 - OS: Ubuntu 22.04 LTS 
 - Thingsboard IoT Gateway version 3.4.5 (and maybe others)
 - Python version: whatever is in the container

To try to troubleshoot this, I pulled tag 3.3 this same system (arm64, ubuntu) and it runs just fine.  I tried pulling 3.4.1 to try to figure out where the problem creeped in, and 3.4.1 has the same exec format error.

wz2b, 2024-03-12:
I do notice this difference in the Dockerfile:

3.3: `FROM python:3.10-slim`
3.4.5: `FROM --platform=$BUILDPLATFORM python:3.11-slim AS build`

I don't know what your CI setup is, but should that be $TARGETPLATFORM ?  Or left unspecified?  What I can't tell is if this is actually a CI problem or this.

fjpanag, 2024-03-23:
The same issue started appearing for me for v3.4.6, on linux/amd64.

The previous version v3.4.5 works fine for me.

imbeacon, 2024-03-25:
Hi, please try now, we have updated docker images on docker hub for 3.4.6.

fjpanag, 2024-03-25:
> Hi, please try now, we have updated docker images on docker hub for 3.4.6.

Hello, I just tried again with v3.4.6 on amd64 and the issue persists.

robjuz, 2024-04-22:
I have the same problem. `3.3` is working. `3.4` and up, no

robjuz, 2024-04-22:
workaround:
1. clone the repo `git clone https://github.com/thingsboard/thingsboard-gateway.git` on the target machine
2. `git checkout 4.3.6`
3. `cd thingsboard-gateway`
4. build the image `docker build -t thingsboard-gateway --file docker/Dockerfile .`
5. use `thingsboard-gateway`

imbeacon, 2024-05-31:
Hi @wz2b,

This issue was solved in the latest version (3.5) please try it and let us know if something goes wrong by opening a new issue.

closed 2024-05-31 by imbeacon