thingsboard/thingsboard-gateway #1335: [HELP] Why are you writing to a config file during runtime

zwoabier, 2024-03-13:
**Describe the issue**
With [this commit](https://github.com/thingsboard/thingsboard-gateway/commit/5576a4115ef6da2cf55e1c90fd103c8f236ec5d1#diff-a91c0a0346854647a41d8662e51ceafdd1a1c1ea11755898a8cdf1eef3cc8598) that was added in version 3.4.5 of the thingsboard gateway you are writing an ID for connectors to the config file during runtime. This requires the config file to be writable.

But especially for kubernetes deployments where you add config files as immutable config maps this change breaks the deployment, because you cannot write to the file.

In general you should maybe discuss if it is a good practice to manipulate user config files during runtime.

If you require an ID for each connector shouldn't it be included in the config given by the user already?
So maybe update default/example bt-gateway config JSON and add the field `id` to the connectors?

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
```json
{
  "thingsboard": {
    "host": "thingsboard.example.com",
    "port": 8883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "tb_gateway_access_token"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 120,
      "inactivityCheckPeriodSeconds": 10
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
    {
      "name": "MQTT Broker Connector",
      "type": "mqtt",
      "configuration": "mqtt.json"
    }
  ]
}

```

**Connector name (If you need help with some connector/converter):**
This affects all connectors but in my case it is the  MQTT Connector

**Error traceback (If it was raised):**
```
2024-03-07 10:45:09 - |ERROR| - [tb_logger.py] - tb_logger - exception - 98 - Error on loading connector: OSError(30, 'Read-only file system')
Traceback (most recent call last):
 File "/thingsboard_gateway/gateway/tb_gateway_service.py", line 818, in _load_connectors
  with open(config_file_path, 'w', encoding="UTF-8") as conf_file:
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 30] Read-only file system: '/thingsboard_gateway/config/mqtt.json'
```

**Versions (please complete the following information):**
 - docker container
 - Thingsboard IoT Gateway version 3.4.5

imbeacon, 2024-03-13:
Hi, you are right, we have added UI configuration and we decided to add identifiers for connectors. Also, you can change the configuration completely from ThingsBoard IoT gateways dashboard. So it is necessary to have ability to write to configuration files now. And now they act not like a constant configuration, but like a persistent configuration files, that will be used if you disconnect the gateway and start it again.
In order to configure the connection you can use environmental variables or general configuration file in persistent volume.
Connectors you can add and manage on UI side.

imbeacon, 2024-03-25:
@zwoabier , Yes, you can set "remoteConfiguration" parameter to False and it won't change the configuration from UI.

closed 2024-04-17 by imbeacon