thingsboard/thingsboard-gateway #1341: [HELP] tb_device_mqtt - _on_connect - 376 - connection FAIL with error Bad user name or password Bad user name or password

Cvd0, 2024-03-19:
**Describe the issue**
Trying to connect the thingsboard IoT gateway to the thingsboard cloud platform. Ill post the error message underneath.

Cannot load GRPC connector!
2024-03-19 12:52:38 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 218 - Gateway starting...
2024-03-19 12:52:38 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 223 - ThingsBoard IoT gateway version: 3.4.5
2024-03-19 12:52:38 - |INFO| - [tb_gateway_service.py] - tb_gateway_service - __init__ - 309 - Gateway started.
2024-03-19 12:52:38 - |INFO| - [tb_loader.py] - tb_loader - import_module - 68 - Import FTPConnector from C:\Python310\lib\site-packages\thingsboard_gateway\connectors\ftp.
2024-03-19 12:52:38 - |INFO| - [ftp_connector.py] - ftp_connector - __init__ - 79 - FTP Connector started.
2024-03-19 12:52:39 - |ERROR| - [tb_device_mqtt.py] - tb_device_mqtt - _on_connect - 376 - connection FAIL with error Bad user name or password Bad user name or password
2024-03-19 12:52:39 - |INFO| - [tb_gateway_mqtt.py] - tb_gateway_mqtt - gw_subscribe_to_attribute - 181 - Subscribed to *|* with id 1 for device *
2024-03-19 12:52:39 - |INFO| - [ftp_connector.py] - ftp_connector - __connect - 127 - Connected to FTP server

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**
`{
  "thingsboard": {
    "host": "thingsboard.cloud",
    "port": 1883,
    "remoteShell": false,
    "remoteConfiguration": true,
    "statistics": {
      "enable": true,
      "statsSendPeriodInSeconds": 3600
    },
    "deviceFiltering": {
      "enable": false,
      "filterFile": "list.json"
    },
    "maxPayloadSizeBytes": 1024,
    "minPackSendDelayMS": 200,
    "minPackSizeToSend": 500,
    "checkConnectorsConfigurationInSeconds": 60,
    "handleDeviceRenaming": true,
    "security": {
      "type": "accessToken",
      "accessToken": "xxxxxx"
    },
    "qos": 1,
    "checkingDeviceActivity": {
      "checkDeviceInactivity": false,
      "inactivityTimeoutSeconds": 200,
      "inactivityCheckPeriodSeconds": 500
    }
  },
  "storage": {
    "type": "memory",
    "read_records_count": 100,
    "max_records_count": 100000,
    "data_folder_path": "./data/",
    "max_file_count": 10,
    "max_read_records_count": 10,
    "max_records_per_file": 10000,
    "data_file_path": "./data/data.db",
    "messages_ttl_check_in_hours": 1,
    "messages_ttl_in_days": 7
  },
  "grpc": {
    "enabled": false,
    "serverPort": 9595,
    "keepaliveTimeMs": 10000,
    "keepaliveTimeoutMs": 5000,
    "keepalivePermitWithoutCalls": true,
    "maxPingsWithoutData": 0,
    "minTimeBetweenPingsMs": 10000,
    "minPingIntervalWithoutDataMs": 5000,
    "keepAliveTimeMs": 10000,
    "keepAliveTimeoutMs": 5000
  },
  "connectors": [
{
      "type": "ftp",
      "name": "FTP Connector",
      "configuration": "ftp.json"
    }
]
}`

**Connector name (If you need help with some connector/converter):**
none

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: windows server 2019
 - Thingsboard IoT Gateway version [3.4.5]
 - Python version[3.10.0]

shanjunying, 2024-03-28:
Hello! I have encountered the same problem. My thingsboard version is 3.6, and this error occurs in versions after using tb-gateway 3.4. Only with tb-gateway version 3.3 can it run properly. How can I solve this problem or can I provide a list of versions corresponding to thingsboard and gateway？Thanks a lot！

Cvd0, 2024-03-28:
@shanjunying Hey, I took an old pc and I'm running an ubuntu server on it. Then again the ubuntu installation guide on the site didn't work for me either. So I did the pip method on the Ubuntu server and now I'm not having any issues (to my knowledge).

shanjunying, 2024-03-29:
Thank you very much for your answer. I will try！

shanjunying, 2024-04-02:
> > Hello! I have encountered the same problem. My thingsboard version is 3.6, and this error occurs in versions after using tb-gateway 3.4. Only with tb-gateway version 3.3 can it run properly. How can I solve this problem or can I provide a list of versions corresponding to thingsboard and gateway？Thanks a lot！
> 
> def get_env_variables(): env_variables = { 'host': environ.get('host'), 'port': int(environ.get('port')) if environ.get('port') else None, 'accessToken': environ.get('accessToken'), 'caCert': environ.get('caCert'), 'privateKey': environ.get('privateKey'), 'cert': environ.get('cert'), 'clientId': environ.get('clientId'), **#'username': environ.get('username'), #注释了这一行，这句是读取环境变量，解决报错（ connection FAIL with error Bad user name or password Bad user name or password）；** 'password': environ.get('password') }

问题已解决！感谢！但是新的问题又出现了，在使用仪表盘的网关连接器时会报错：
|ERROR| - [tb_logger.py] - tb_logger - exception - 98 - 'deviceType'
Traceback (most recent call last):
  File "C:\thingsboard-gateway\thingsboard_gateway\gateway\tb_gateway_service.py", line 1547, in __load_persistent_devices
    self.add_device(device, self.__connected_devices[device], self.__connected_devices[device][
KeyError: 'deviceType'

这个问题您有什么思路吗

samson0v, 2024-06-11:
Hi @Cvd0

This issue is fixed in the master branch, you can use it or wait for the release.

closed 2024-06-19 by imbeacon