thingsboard/thingsboard-gateway #1352: [BUG] Architecture issue with latest tb-gateway image (3.4.6)

DigitalPals, 2024-03-24:
**Docker Version:** 26.0.0, build 2ae903e

**Operating System:** Ubuntu 22.04 LTS x86_64

**Description:**
When running docker compose up with a Docker Compose file, I'm encountering an "exec format error" when attaching to one of the containers. The container exits with code 0 immediately after starting.

**Steps to Reproduce:**
Create a new directory with a docker-compose.yaml file
Add a service definition like:

```
services:
  # ThingsBoard IoT Gateway Service Configuration
  tb-gateway:
    image: thingsboard/tb-gateway
    container_name: tb-gateway
    restart: always
```

Run docker compose up

**Expected Behavior:**
The container should start successfully and attach without any errors.

**Actual Behavior:**
```
Attaching to tb-gateway
myservice  | exec /bin/sh: exec format error
myservice exited with code 0
```

**Additional Context:**
The Docker image tb-gateway:latest has an architecture of "amd64"

The host system is x86_64, which should be compatible with amd64 images

This issue does not occur on older tb-gateway version (e.g. 3.4.5)

Let me know if you need any other information to investigate this issue further.

imbeacon, 2024-03-25:
Hi @DigitalPals,

We have updated images on docker hub, please try now.

DigitalPals, 2024-03-25:
Hi @imbeacon, I can confirm it starts now. It keeps saying that there is a new version available though. I double checked with `docker image inspect 2523b21371ba` it was created today. 

```
[===UPDATE===]

 New version 3.4.6 is available! 

[===UPDATE===]
```

samy4sam, 2024-03-26:
@DigitalPals 
The new version message is another bug: [https://github.com/thingsboard/thingsboard-gateway/issues/1355](https://github.com/thingsboard/thingsboard-gateway/issues/1355)

DigitalPals, 2024-03-26:
@samy4sam Thanks, I'll close this issue then.

closed 2024-03-26 by DigitalPals