thingsboard/thingsboard-gateway #1354: [BUG] OPC-UA AsyncIO does not find sub variables

samy4sam, 2024-03-25:
**Describe the bug**
The OPC-UA AsyncIO connector does not find sub variables like the OPC-UA Connector.

- Old issue: [https://github.com/thingsboard/thingsboard-gateway/issues/1249](https://github.com/thingsboard/thingsboard-gateway/issues/1249)

- Old Pull request: [https://github.com/thingsboard/thingsboard-gateway/pull/1251](https://github.com/thingsboard/thingsboard-gateway/pull/1251) 

**Connector name (If bug in the some connector):**
OPC-UA AsyncIO Connector

**Versions (please complete the following information):**
 - OS: Docker
 - Thingsboard IoT Gateway version: 3.4.6

samy4sam, 2024-04-30:
@samson0v @imbeacon any news on this bug?

samson0v, 2024-05-20:
Hi @samy4sam,
I tried to reproduce your issue but I can't.
I have the following OPC-UA structure:
[image]

I trying to read **CounterSubVar**, for this purpose I used the following connector config:
```json
"mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Simulation",
        "deviceNamePattern": "Device asd",
        "attributes": [],
        "timeseries": [
          {
            "key": "humidity_sub",
            "path": "${Test\\.Counter\\.CounterSubVar}"
          }
        ],
      ....
```

And as you can see, sub variable read successfully:
[image]

closed 2024-06-11 by samy4sam