thingsboard/thingsboard-gateway #1358: [HELP] modbus TCP reads only first "timeseries"

ilpaso, 2024-03-29:
**Describe the issue**
I'm trying to read 3 variables and an attribute from a Modbus TCP device.
The tb_gateway reads only the first variable. I think there is a timing issue because the slave device needs a delay of about 150ms between 2 readings.
I wrote a simple python code to read the 3 variables and it works. 
Thank you for your support

**Connector name (If you need help with some connector/converter):**
MODBUS Connector (modbus.json)

{
  "master": {
    "slaves": [
      {
        "host": "192.168.0.233",
        "port": 8899,
        "type": "tcp",
        "method": "socket",
        "timeout": 1,
        "byteOrder": "BIG",
        "wordOrder": "BIG",
        "retries": true,
        "retryOnEmpty": true,
        "retryOnInvalid": true,
        "pollPeriod": 5000,
        "unitId": 1,
        "deviceName": "energy_meter_1",
        "sendDataOnlyOnChange": false,
        "connectAttemptTimeMs": 5000,
        "connectAttemptCount": 5,
        "waitAfterFailedAttemptsMs": 10000,
        "attributes": [
          {
            "tag": "serial_number",
            "type": "64uint",
            "functionCode": 3,
            "objectsCount": 4,
            "address": 0
          }
        ],
        "timeseries": [
          {
            "tag": "L1_voltage",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 14
          },
          {
            "tag": "L2_voltage",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 16
          },
          {
            "tag": "L3_voltage",
            "type": "32float",
            "functionCode": 3,
            "objectsCount": 2,
            "address": 18
          }
        ],
        "attributeUpdates": [],
        "rpc": []
      }
    ]

**Versions**
 - OS:  Raspbian GNU/Linux 11 (bullseye)
 - Thingsboard IoT Gateway version: 3.4.6
 - Python version: 3.9.2

samson0v, 2024-06-11:
Hi @ilpaso, thanks for your interest in ThingsBoard IoT Gateway!
Please, send the gateway logs file.

imbeacon, 2024-10-14:
Hi,

The issue is not reproducible in the latests release, please try to use it and let us know if you still encounter an issue.

closed 2024-10-14 by imbeacon