thingsboard/thingsboard-gateway #1360: Modbus|ERROR| - [tb_logger.py] - tb_logger - exception - 98 - Exception Response(131, 3, IllegalFunction)[HELP]

DuXiaoChuang, 2024-04-01:
**Describe the issue**
 |ERROR| - [tb_logger.py] - tb_logger - exception - 98 - Exception Response(131, 3, IllegalFunction)

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

mrangics, 2024-04-05:
Hi,

this means that you do not have such a register 62363. You have to check with a ModBus Reader like qModBus or something else, which is the correct register to read.
I think you should also check if you are reading the other values correctly, using the correct data type as I'm suspecting, that the 32int is not the correct data type, maybe 16uint or 32uint but a 32int is definitely surprising.

DuXiaoChuang, 2024-04-07:
> Hi,
>
> this means that you do not have such a register 62363. You have to check with a ModBus Reader like qModBus or something else, which is the correct register to read. I think you should also [...]

ok，thank you ! you can close it .

closed 2024-05-31 by imbeacon