thingsboard/thingsboard-gateway #1361: [BUG] 100% CPU usuage of thingsboard-gateway

toml55, 2024-04-03:
Hi,
I have the same issue like these ones with version thingsboard-gateway 3.4.6:
https://github.com/thingsboard/thingsboard/issues/8867
https://github.com/thingsboard/thingsboard-gateway/issues/272

from top output:

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                 
  14351 thingsb+  20   0 1663272  47608  16816 S 100,7   0,8   0:12.78 python3 

(14351 is the thingsboard-gateway task):
thingsb+   14351  100  0.8 1665320 49192 ?       Ssl  20:16  10:41 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Tried to disable all connectors but python3 load stays at ~100%.
Image is the .deb from
https://github.com/thingsboard/thingsboard-gateway/releases/latest/download/python3-thingsboard-gateway.deb
from last week on an ubuntu server.

Versions:
- thingsboard-gateway 3.4.6
- thingsboard 3.6.2
- Ubuntu server 20.04 LTS 
- Python 3.10.12

any possibility to debug this?

Kevin-ox, 2024-04-16:
I have also encountered this issue. The newly installed gateway may have this issue, so I suspect it is related to an incompatible package upgrade

Kevin-ox, 2024-04-17:
Found the problem, it's tb_mqtt_client

ERROR: tb-mqtt-client 1.9.0 has requirement requests>=2.31.0, but you'll have requests 2.22.0 which is incompatible.

befantasy, 2024-05-10:
> Found the problem, it's tb_mqtt_client
> 
> ERROR: tb-mqtt-client 1.9.0 has requirement requests>=2.31.0, but you'll have requests 2.22.0 which is incompatible.

hi, Kevin:
Do you know how can I solve this problem? Upgrade the request module in tb-gateway to the latest version?

befantasy, 2024-05-10:
Roll back to version 3.44, CPU usage returned to normal.

imbeacon, 2024-06-06:
Hi @befantasy,

The correct way to solve this - install requests versions >= 2.31.0. It should install it automatically, but by some reason it didn't do this, we will check this point.

closed 2024-06-19 by imbeacon