thingsboard/thingsboard-gateway #1362: When The gateway runs for a long time, RuntimeError: Message publish failed: The client is not currently connected.

DuXiaoChuang, 2024-04-07:
**Describe the issue**
When The gateway runs for a long time, RuntimeError: Message publish failed: The client is not currently connected.
2024-04-07 09:26:54 - |ERROR| - [tb_logger.py] - tb_logger - exception - 97 - Message publish failed: The client is not currently connected.
Traceback (most recent call last):
  File "C:\code\longchengcloud-gateway\thingsboard_gateway\gateway\tb_gateway_service.py", line 1134, in __read_data_from_storage
    success = event.get() == event.TB_ERR_SUCCESS
              ^^^^^^^^^^^
  File "C:\code\longchengcloud-gateway\venv\Lib\site-packages\tb_device_mqtt.py", line 179, in get
    self.message_info.wait_for_publish(timeout=1)
  File "C:\code\longchengcloud-gateway\venv\Lib\site-packages\paho\mqtt\client.py", line 362, in wait_for_publish
    raise RuntimeError('Message publish failed: %s' % (error_string(self.rc)))
RuntimeError: Message publish failed: The client is not currently connected.

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you want to attach a tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. Modbus Connector]

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.0.1]
 - Python version[e.g. 3.9]

imbeacon, 2024-06-19:
Hi @DuXiaoChuang,

It is possible to get this error, it means that connection was lost, but the gateway will resend the event pack.

closed 2024-06-19 by imbeacon

DuXiaoChuang, 2024-06-24:
> Hi @DuXiaoChuang,
> 
> It is possible to get this error, it means that connection was lost, but the gateway will resend the event pack.

how to reslove this problem