thingsboard/thingsboard-gateway #1371: Cannot load GRPC connector！Gateway Shell

ZhaoLinl, 2024-04-17:
Thingsboard IoT Gateway version [3.4.6] An error occurred after the IDEA was imported and the program was running.
[image]

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [3.4.6]
 - Python version[3.9]

ZhaoLinl, 2024-04-18:
Can you help me with this problem?

samson0v, 2024-05-28:
Hi @ZhaoLinl, thanks for your interest in ThingsBoard IoT Gateway!
"Cannot load GRPC connector!" - it is not an error and it is normal behaviour.

Can you provide a full backtrace of the second error, I cannot understand what caused this?

ZhaoLinl, 2024-06-06:
I have understood the question, thank you for your answer.

closed 2024-06-19 by imbeacon

lolPlatinumPlayer, 2024-08-12:
> Hi @ZhaoLinl, thanks for your interest in ThingsBoard IoT Gateway! "Cannot load GRPC connector!" - it is not an error and it is normal behaviour.
> 
> Can you provide a full backtrace of the second error, I cannot understand what caused this?

I dont understand why it is normal behaviour. [ @samson0v 's comment in issue1341](https://github.com/thingsboard/thingsboard-gateway/issues/1341#issuecomment-2160177531) seem like this issue is fixed in the master branch. But whether I run it with docker or with source code of the master branch, I can still see `Cannot load GRPC connector!`. (I use Windows 11 and python 3.12.3)