thingsboard/thingsboard-gateway #1375: Can't correctly setup the rest.json configuration file [HELP]

IhorYednak, 2024-04-22:
**Description**
Hello, I'm trying to configure and test the rest.json connector. I want to send a post request using a postman.
to see some data coming through, so I got a new device created in Thingsboard based on the Post request.

**Configuration**
[rest.json](https://github.com/thingsboard/thingsboard-gateway/files/15065240/rest.json)

config file is just basic config
maybe I'm missing something in host, and there should be another IP (cuz it's default) like the IP of the VM where gateway is running, or the address of the thingsboard instance.
authentication is anonymous

**Connector name**
[e.g. REST Connector]

**Error traceback :**
Error from Postman when I click send
**POST http://92.111.176.157:5000/
Error: connect ECONNREFUSED 92.111.176.157:5000
Request Headers
Content-Type: application/json
Authorization: Basic Og==
User-Agent: PostmanRuntime/7.37.3
Accept: */*
Postman-Token: 188e6235-9496-4c0f-a128-34e4f105e007
Host: 92.111.176.157:5000
Accept-Encoding: gzip, deflate, br
Connection: keep-alive
Request Body
{
	"name": "Ihor's Test",
    "parameterOne": "Value1",
    "parameterTwo": "Value2"
}
**
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [Ubuntu server 20.04]
 - Thingsboard IoT Gateway version [e.g. 3.4.6]
 - Python3 version[ 3.10.12]

Thanks in advance!

IhorYednak, 2024-04-24:
Thank you for your response @imbeacon this is very useful, I corrected my config file, so now it looks like this:

`"host": "127.0.0.1",
  "port": "5000",
  "SSL": false,
  "mapping": [
    {
      "endpoint": "/my_endpoint",
      "HTTPMethods": [
        "POST"
      ],
      "security": {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "${name}",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "firstParameter",
            "value": "${parameterOne}"
          }
        ],
        "timeseries": [
          {
            "type": "string",
            "key": "secondParameter",
            "value": "${parameterTwo}"
          }
        ]
      }
    },`

But still I got error in Postman and can't process the request
[image]

Maybe there's something to do with the URL ip address?

IhorYednak, 2024-04-24:
@imbeacon 
This is what I got 
[image]

IhorYednak, 2024-04-24:
[image]
It should be good I think

imbeacon, 2024-04-24:
Ok, it looks like the request cannot arrive to the server, please try to run the following command from the server:

```
curl -X POST -H "Content-Type: application/json" -d '{
    "sensorName": "Ihor'\''s Test",
    "sensorType": "default",
    "sensorModel": "Value1",
    "certificateNumber": "Value2"
}' http://127.0.0.1:5000/my_devices
```

and from external machine:
```
curl -X POST -H "Content-Type: application/json" -d '{
    "sensorName": "Ihor'\''s Test",
    "sensorType": "default",
    "sensorModel": "Value1",
    "certificateNumber": "Value2"
}' http://92.111.176.157:5000/my_devices
```

If first arrives - the issue may be in firewall or network, for example with port forwarding.

IhorYednak, 2024-04-24:
Thank you @imbeacon, the issue is solved, I was able to get the data over it by typing the address of the VM itlsef in the rest.json file instead of 127.0.0.1 and after that post data from external machine.
so 5 IOT starts of 5 😄

closed 2024-04-25 by IhorYednak