thingsboard/thingsboard-gateway #1376: How to connect my gateway to thingsboard Community Edition(Thingsboard CE)

kookohd, 2024-04-23:
I am working on a project to send data from lorawan devices to thingsboard CE. As you may already have known lorawan devices are low power and long range. therefore they need the help of lora gateway to connect to thingsboard CE.
So far what i have done is get my network server up, connecting my lorawan devices to my network server.
After that i set up the thingsboard server and check connectivity and everything is ok. the problem that i face is that I am unsure how to connect the gateway to thingsboard to allow the data i have collected to reach thingsboard CE to receive my data and put it in a dashboard.

What i have done so far. I have looked at a few documentation on mqtt api ref, rule engine and connectors. However i am not sure how  to put all these information together to obtain what i need. The mqtt broker i am currently running is the one that thingsboard CE originally has.

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 22.04] <- this is where my thingsboard server is located
 - Thingsboard CE version 3.6.4
 - Wisgate OS RAK7268 as my gateway.

kookohd, 2024-04-23:
@imbeacon and @samson0v, not to be rude. But if you could really help me out on this, it will be very life changing. I have noticed that there is quite a few questions that have been opened but not answered. thank you very much for your time.

imbeacon, 2024-04-24:
Hi @kookohd,

Thank you for your interest in Thingsboard IoT Gateway. Actually, usually network servers provides setup with mqtt brokers that you can run and get a data using them, for example Chirpstack, where you can run mosquitto broker in default docker-compose setup. To connect to it and get a data you can use MQTT connector of the Thingsboard IoT Gateway or MQTT integration in ThingsBoard PE. 
You can find information about on how to configure the connector in [our official documentation](https://thingsboard.io/docs/iot-gateway/).
The configuration very depends on your application server configuration and its integration ability. You can find some examples on how to connect LoRaWAN devices in [our devices library](https://thingsboard.io/docs/devices-library/), for example - [article about the LoRaWAN gateway connection](https://thingsboard.io/docs/devices-library/ug56-lorawan-gateway/) and [article about connection device through connected earlier LoRaWAN gateway](https://thingsboard.io/docs/devices-library/ws202-pir-and-light-sensor/). These article uses integrations, but you can configure the gateway to get a data and send it to ThingsBoard.

closed 2024-05-28 by imbeacon