thingsboard/thingsboard-gateway #1387: [HELP] Is there an easy way to read same timeseries from multiple devices Modbus TCP?

franklupo, 2024-05-03:
I need to read about 25 timeseries data from the device and I need more than 300 devices to configure.
Modbus.json too large to maintain.

How can I do?

Best regards

mrangics, 2024-05-17:
Unfortunately you'll have to write a script for this handing it.
Be careful, the Modbus connector is extremely full of bugs, if you need to read the timeseries data for multiple registrers, it doesn't work. I'm heavily suggesting swapping to another method.
We also started with the ModBus connector, but ended up using the Mikrotik KNOT device and MQTT. It works, it's stable and easy to scale.

imbeacon, 2024-06-19:
Hi, unfortunately, at the moment there is no option to do this, but we are working on UI side configuration and it will be useful in cases like you described.

closed 2024-06-19 by imbeacon