thingsboard/thingsboard-gateway #1396: [Modbus RTU error] Port [ttyUSB0]  Permission denied: '/dev/ttyUSB0'

pro100805, 2024-05-13:
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-05-13 09:19:30 CEST; 21s ago
   Main PID: 16644 (python3)
      Tasks: 21 (limit: 2219)
     Memory: 58.8M
        CPU: 4.642s
     CGroup: /system.slice/thingsboard-gateway.service
             └─16644 /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

May 13 09:19:44 tb-gw-server python3[16644]: 2024-05-13 09:19:44 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:44 tb-gw-server python3[16644]: 2024-05-13 09:19:44 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:44 tb-gw-server python3[16644]: 2024-05-13 09:19:44 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'
May 13 09:19:49 tb-gw-server python3[16644]: 2024-05-13 09:19:49 - |ERROR| - [serial.py] - serial - connect - 221 - [Errno 13] could not open port /dev/ttyUSB0: [Errno 13] Permission denied: '/dev/ttyUSB0'

imbeacon, 2024-05-13:
Hi @pro100805,

Please try to add user thingsboard_gateway to group dial out and restart a machine.

To do this you can use the following command:
`sudo adduser thingsboard_gateway dialout`

pro100805, 2024-05-13:
good afternoon, after the operations performed the following message appears

● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-05-13 10:20:53 CEST; 19min ago
   Main PID: 1439 (python3)
      Tasks: 20 (limit: 2219)
     Memory: 48.0M
        CPU: 2min 48.688s
     CGroup: /system.slice/thingsboard-gateway.service
             └─1439 /usr/bin/python3 -c "from thingsboard_gateway.tb_gateway import daemon; daemon()"

May 13 10:21:47 tb-gw-server python3[1439]:     self.run()
May 13 10:21:47 tb-gw-server python3[1439]:   File "/usr/lib/python3.10/threading.py", line 953, in run
May 13 10:21:47 tb-gw-server python3[1439]:     self._target(*self._args, **self._kwargs)
May 13 10:21:47 tb-gw-server python3[1439]:   File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 422, in __process_slaves
May 13 10:21:47 tb-gw-server python3[1439]:     self.__gateway.del_device(device.device_name)
May 13 10:21:47 tb-gw-server python3[1439]:   File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 1453, in del_device
May 13 10:21:47 tb-gw-server python3[1439]:     self.__save_persistent_devices()
May 13 10:21:47 tb-gw-server python3[1439]:   File "/usr/local/lib/python3.10/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 1569, in __save_persistent_devices
May 13 10:21:47 tb-gw-server python3[1439]:     with open(self._config_dir + CONNECTED_DEVICES_FILENAME, 'w') as config_file:
May 13 10:21:47 tb-gw-server python3[1439]: PermissionError: [Errno 13] Permission denied: '/etc/thingsboard-gateway/config/connected_devices.json'

pro100805, 2024-05-13:
via tcp everything works without problems

pro100805, 2024-05-13:
Good evening, everything works fine, thanks for the quick help :)

pro100805, 2024-05-16:
Good afternoon, when working through the serial port there is a problem in communication, it is related to RPC requests - responses and manifests itself as follows
- if no queries are made, everything works stably, telemetry is received without problems or delays
- after several RPC requests (possibly more than 5) the connection is disconnected for a while and then restored, and this situation is cyclic.
- the same equipment with the same settings via TCP works stably
= please help me
What information do I need to provide to help you solve this problem?

closed 2024-05-28 by imbeacon