thingsboard/thingsboard-gateway #1402: Issue while installing Thingsboard IoT Gateway on Raspberry Pi OS [HELP]

shreyasnmhatre, 2024-05-16:
**Describe the issue**
I am currently attempting to install thingsboard IoT gateway on a Raspberry Pi by following the Debian installation instructions.

After issuing sudo apt install ./python3-thingsboard-gateway.deb -y I'm getting the following message:

[image]

**Versions (please complete the following information):**
-OS: [Raspbian GNU (Bullseye)]
-Thingsboard IoT Gateway version [3.1]
-Python version[3.9.2]

tatuiot, 2024-06-15:
Same issue here , managed to install via pip but I am facing these issues
1. Cannot make it log locally - error Logging loading exception, logs.json is wrong: Unable to configure handler 'connectorHandler' 
2. Cannot send data over MQTT  (added a MQTT connector) and configured mqtt.json file , local mosquitto is working accepting connections locally and remotly. there is no error but when sending data .... nothing is display in TB Device console. 

I'll keep trying...  but so far 2 days o learning and testing... no success. thanks

cos-gl, 2024-06-18:
Thansk @tatuiot for your feedback!
> Same issue here , managed to install via pip but I am facing these issues
Which version of python3-cryptography do you installed with pip?

For now I've gived up and installed TB-Gateway wth docker (but was not my preferred solution in this case).
If you have news I will be happy to hear about

cos-gl, 2024-06-21:
Hi @shreyasnmhatre the dependency problem seems to be fixed with the latest release of gateway: 3.5.1-1

closed 2024-07-01 by imbeacon