thingsboard/thingsboard-gateway #1406: [Modbus RTU error] when working through the serial port

pro100805, 2024-05-23:
Good afternoon, when working through the serial port there is a problem in communication, it is related to RPC requests - responses and manifests itself as follows

    if no queries are made, everything works stably, telemetry is received without problems or delays
    after several RPC requests (possibly more than 5) the connection is disconnected for a while and then restored, and this situation is cyclic.
    the same equipment with the same settings via TCP works stably
    = please help me
    What information do I need to provide to help you solve this problem?

imbeacon, 2024-06-06:
Hi @pro100805,

This issue was solved in the master branch and fix will be available in the nearest release, or you may use the version from the master branch.

pro100805, 2024-06-06:
please tell me how I can use the master branch version.

imbeacon, 2024-06-19:
We have released a new version, you can just use it.

closed 2024-06-19 by imbeacon